SBFspot/SBFspot #336: Sunny Boy Storage 2.5 not correct detected

anwelti, 2019-11-08:
Hello and sorry for my poor english,

i think my Sunny Boy Storage 2.5 is not correct detect as an BatteryInverter. in my iobroker i cant see the battery charging status but grid status.

is this an bug or just bad config?

Thx
Andi

sbf-, 2019-11-08:
> is this an bug or just bad config?

Can't tell without some more info...
What's in your config?
Can you post the output of `SBFspot -v5 -finq`

anwelti, 2019-11-11:
okay, sorry, i give u a little feed:

this is output of our command (without SN):
SBFspot V3.6.0
Yet another tool to read power production of SMA solar inverters
(c) 2012-2019, SBF (https://github.com/SBFspot/SBFspot)
Compiled for Linux (LE) 32 bit with SQLite support
Commandline Args: -v5 -finq
Reading config '/usr/local/bin/sbfspot.3/SBFspot.cfg'
Configuration settings:
IP_Address=192.168.188.31
Password=<undisclosed>
MIS_Enabled=1
Plantname=MyPlant
OutputPath=/home/pi/smadata/%Y
OutputPathEvents=/home/pi/smadata/%Y/Events
Latitude=52.52
Longitude=13.4
Timezone=Europe/Berlin
CalculateMissingSpotValues=0
DateTimeFormat=%d/%m/%Y %H:%M:%S
DateFormat=%d/%m/%Y
TimeFormat=%H:%M:%S
SynchTime=1
SynchTimeLow=1
SynchTimeHigh=3600
SunRSOffset=900
DecimalPoint=comma
CSV_Delimiter=semicolon
Precision=3
CSV_Export=1
CSV_ExtendedHeader=1
CSV_Header=1
CSV_SaveZeroPower=1
CSV_Spot_TimeSource=0
CSV_Spot_WebboxHeader=0
Locale=en-US
BTConnectRetries=10
SQL_Database=/home/pi/smadata/SBFspot.db
### End of Config ###
Mon Nov 11 13:05:02 2019: INFO: Starting...
sunrise: 07:21
sunset : 16:18
Connecting to Local Network...
Initializing...
SUSyID: 125 - SessionID: 966428040
Inverter IP address: 192.168.188.31 from SBFspot.cfg
Inverter replied: 192.168.188.31 SUSyID: 346 - Serial: 
Inverter IP address: 192.168.188.12 from SBFspot.cfg
Inverter replied: 192.168.188.12 SUSyID: 378 - Serial: 
Logon OK
SUSyID: 346 - SN: 
Device Name:      SN: 
Device Class:     Battery Inverter
Device Type:      Sunny Boy Storage 2.5
Software Version: 02.04.24.R
Serial number:    
SUSyID: 378 - SN: 
Device Name:      SN: 
Device Class:     Solar Inverters
Device Type:      Sunny Tripower 5.0
Software Version: 02.13.33.R
Serial number:    
SUSyID: 346 - SN: 
Batt. Charging Status: 0%
SUSyID: 346 - SN: 
Batt. Temperature: 20.0°C
Batt. Voltage    : 1.14V
Batt. Current    : 0.000A
SUSyID: 346 - SN: 
Grid Power Out : 0W
Grid Power In  : 0W
SUSyID: 346 - SN: 
Device Status:      Ok
SUSyID: 378 - SN: 
Device Status:      Ok
SUSyID: 346 - SN: 
Device Temperature: 0.0°C
SUSyID: 378 - SN: 
Device Temperature: 0.0°C
SUSyID: 346 - SN: 
Pac max phase 1: 2500W
Pac max phase 2: 0W
Pac max phase 3: 0W
SUSyID: 378 - SN: 
Pac max phase 1: 5000W
Pac max phase 2: 0W
Pac max phase 3: 0W
SUSyID: 346 - SN: 

this is my config (without passwd):
#
# SBFspot.cfg - Created Mi 30. Okt 23:04:32 CET 2019 with sbfspot-config V1.0.15
# For detailed configuration info, refer to SBFspot.default.cfg
#
MIS_Enabled=1
IP_Address=192.168.188.31,192.168.188.12
Password=
Plantname=MyPlant
OutputPath=/home/pi/smadata/%Y
OutputPathEvents=/home/pi/smadata/%Y/Events
CSV_Export=1
CSV_ExtendedHeader=1
CSV_Header=1
CSV_SaveZeroPower=1
CSV_Delimiter=semicolon
CSV_Spot_TimeSource=inverter
CSV_Spot_WebboxHeader=0
Timezone=Europe/Berlin
Latitude=52.52
Longitude=13.40
Locale=en-US
DateTimeFormat=%d/%m/%Y %H:%M:%S
DateFormat=%d/%m/%Y
TimeFormat=%H:%M:%S
DecimalPoint=comma
CalculateMissingSpotValues=0
SQL_Database=/home/pi/smadata/SBFspot.db
# End of Config

And this is the output in iobroker:
[image]

I think for battery inverter i must see other values, like storage charging state or else.

closed 2021-04-18 by sbf-