SBFspot/SBFspot #339: Random 0.00V Uac and Iac values

scotte, 2019-11-15:
About once a day, SBFspot scrapes my SB7.7-1SP-US-40 and gets 0.00V for Uac and Iac values. All other readings, such as generation values and Udc/Idc values are fine.

Example normal log output:

```
AC Spot Data:
        Phase 1 Pac :   2.842kW - Uac: 125.20V - Iac: 22.727A
        Phase 2 Pac :   2.831kW - Uac: 124.74V - Iac: 22.727A
        Phase 3 Pac :   0.000kW - Uac:   0.00V - Iac:  0.000A
        Total Pac   :   5.673kW
```

Abnormal log output:
```
AC Spot Data:
        Phase 1 Pac :   2.850kW - Uac:   0.00V - Iac:  0.000A
        Phase 2 Pac :   2.843kW - Uac:   0.00V - Iac:  0.000A
        Phase 3 Pac :   0.000kW - Uac:   0.00V - Iac:  0.000A
        Total Pac   :   5.694kW
```

There's nothing else interesting in the logs that I can see, and as I mentioned above, the DC values are fine, as is AC kW generation.

This really isn't a big deal at all, the Uac/Iac information isn't critical, but I thought I would report this anyway.

sbf-, 2019-11-22:
How is the connection? Bluetooth/Wifi/wired?
Is the connection stable?
I have a BT connected inverter and sometimes, when the signal is very weak, I have the same issue...

scotte, 2019-11-25:
Interesting, I wouldn't have pegged it as a connection issue as it's only these values that are 0 - I've never seen production, DC, or any other values affected.

The inverter is on wifi, with an OK (not great) signal. I have seen the inverter drop off wifi on rare occasion.

Perhaps this is a case where the call to the inverter could use a retry (if one doesn't already exist, I haven't actually looked)?

scotte, 2020-03-26:
Closing - took a closer look at when this happens, and indeed it's poor connectivity with the inverter (wifi, in my case). Generation data seems to always come through eventually - not sure if that's historical where the instant values are not, or what, but this is certainly not an SBFspot bug from what I can tell.

closed 2020-03-26 by scotte