SBFspot/SBFspot #340: raspberry pi zero w bluetooth stops working after some time

DannyDeGaspari, 2019-12-01:
I don't know if this issue is related to SBFspot. I used SBFspot on a raspberry pi 3b+ for many months without problems. But recently I changed this setup and used a raspberry pi zero w (wifi + bluetooth integrated) and bluetooth stops working after some time (1 or 2 days). SBFspot can't make a connection anymore, also 'hcitool scan' does not show any devices. I'm not able to reset the bluetooth stack, the only thing that helps is rebooting the pi.
I'm using raspbian buster lite (sep 2019) and SBFspot 3.6, also tried v3.5.2. The distance to the SMA converters (2 in my setup) is only 1 m. Also the pi is connected to the network via wifi. Is there anything I can do to make the bluetooth stable or do a bluetooth reset every hour or so ?

stripwax, 2019-12-01:
If it helps (maybe it does not....) I have been using a raspberry pi zero w
for sometime (nearly a year) with no major problems. I think it lost
Bluetooth connection just once. It's left on 24x7 and like yours is
connected to my network via wifi.
It might be a configuration issue, firmware or driver version, or different
behaviour in SBFspot . It's always hard to pinpoint a specific problem (and
therefore a fix) but the hardware seems fine so it's usually a software
issue. Or it could be possible that your pi zero has a hardware problem
unrelated to SBFspot (were you using your pi zero successfully previously
or is this the first usage of it?). Running sudo apt-get update and sudo
apt-get upgrade, at a minimum, might simply resolve if it's a
firmware/driver/configuration problem.

stripwax, 2019-12-01:
Ah, I am using Stretch though, not Buster - it should work with both of
course.

DannyDeGaspari, 2019-12-01:
Thanks, I'll try the full-upgrade option. I have the zero already since the very beginning it came out. But had no long term bluetooth projects until now.

DannyDeGaspari, 2019-12-15:
The (full) upgrade did not help. I reinstalled a fresh Raspbian Stretch from april 2019, and now it is already working without any problem for 2 weeks.
I suspect that the bluetooth is not yet stable enough in Raspbian Buster.

closed 2019-12-16 by DannyDeGaspari