SBFspot/SBFspot #341: Database purge if older than x days?

KiavashS, 2019-12-03:
Hi, I've just noticed that my Windows install on SQLite DB is over 3.5Gb in size. Performance is horrible and takes several minutes to run my view.

Is there a clean up option to have it automatically delete any data older than X days or something?

sbf-, 2019-12-03:
Which view?
Maybe [this](https://github.com/SBFspot/SBFspot/issues/283) can help?

KiavashS, 2019-12-03:
I have a couple of custom views with joins. Actually I can optimise the selects to improve performance, but either way, the DB continuously grows as more samples are taken... I'm using my implementation with my home automation, so sampling pretty much every 5 seconds which over 6 months has grown to 3.5Gb already. I think a purge feature would be quite useful for long life systems.

sbf-, 2019-12-03:
> SQLite DB is over 3.5Gb

what did you do with it? Mine is 65Mb with over 5 years of data.
Did you try the [vacuum](https://www.sqlitetutorial.net/sqlite-vacuum/) command?

KiavashS, 2019-12-03:
Oh that's interesting... I've only added a couple of new views.
My SpotData table contains 4.3M records
SpontDataX contains 13.6M records

You made me have another closer look and it turns out it's the WAL file that's the main problem. I must have just sized the whole folder last time. 

But **SBFspot.db** is still 750Mb which is significantly bigger than your 65Mb.

sbf-, 2020-01-06:
> You made me have another closer look and it turns out it's the WAL file that's the main problem

Have a look at [Write-Ahead Logging](https://www.sqlite.org/wal.html#avoiding_excessively_large_wal_files)
> **Avoiding Excessively Large WAL Files**
So in the vast majority of cases, applications need not worry about the WAL file at all. SQLite will automatically take care of it. But it is possible to get SQLite into a state where the WAL file will grow without bound, causing excess disk space usage and slow queries speeds

sbf-, 2020-06-19:
SQLite query to delete older data:

Select statement to verify (be careful with date format):
``` SQL
select * from daydata
where date(TimeStamp, 'unixepoch', 'localtime') < '2020-01-01';
```
Actual delete query:
``` SQL
delete from daydata
where date(TimeStamp, 'unixepoch', 'localtime') < '2020-01-01';
```
The same can be done with other tables

closed 2020-06-19 by sbf-