SBFspot/SBFspot #342: Table and field descriptions

KiavashS, 2019-12-03:
Hi, I'm looking for descriptions of the various tables and fields in the DB. I'm trying to extract specific information from the DB but the views and tables are not very self explanatory. Some views that combine something like "Pac1+Pac2+Pac3 AS PacTot". Could we have a description of the tables and their fields? That way we could build views or look up data specifically for our own scenario.

Some fields make sense by the name but many are just some letters which then have to be combined to give complete information:

	Pdc1,Pdc2,
	Idc1,Idc2,
	Udc1,Udc2,
	Pac1,Pac2,Pac3,
	Iac1,Iac2,Iac3,
	Uac1,Uac2,Uac3,
	Pdc1+Pdc2 AS PdcTot,
	Pac1+Pac2+Pac3 AS PacTot,
	CASE WHEN Pdc1+Pdc2 = 0 THEN
	    0
	ELSE
	    CASE WHEN Pdc1+Pdc2>Pac1+Pac2+Pac3 THEN
	        ROUND(1.0*(Pac1+Pac2+Pac3)/(Pdc1+Pdc2)*100,1)
	    ELSE
	        100.0
	    END
	END AS Efficiency,

sbf-, 2019-12-03:
If you learn something about electricity, you will see the field names are very self-explaining.
https://en.m.wikipedia.org/wiki/Alternating_current

stripwax, 2019-12-03:
That might be partially true only. Ac=AC and Dc=DC, and P is Power and I is
current. What's U (as in Udc1)? What's the significance of the two DC
values and 3 AC values? I think the original question is still reasonable
to ask.

sbf-, 2019-12-03:
U is voltage (in Europe, but it seems to be V in other parts of the world)
It's to avoid such things as `V=240V` (can be seen as an incorrect mathematical equation)
dc1-2 is for input string 1-2 ([there are a few 3 string inverters too](https://github.com/SBFspot/SBFspot/pull/315))
ac1-3 are the 3 phases
As you can see, there is some logic behind ;-)

KiavashS, 2019-12-08:
Are the AC values mainly related to grid connect systems? In my off grid system (sunny boy to sunny island) I'm missing some values and can't make any sense of it. 
* Pac1, 2 and 3 are all 0.
* Iac1 and Uac1 have values
Should I be expecting a value for Pac1?

sbf-, 2019-12-09:
Pac1 should be visible I guess... does it show up in SunnyExplorer?
There is an option in SBFspot config file: CalculateMissingSpotValues=1

closed 2020-04-27 by sbf-