SBFspot/SBFspot #343: Signed vs Unsigned bug found with high serial number inverters

bshaffer82, 2019-12-13:
I've got a definite bug but I need help finding the right file.

I've been running SBFspot for years and I recently got two new inverters and they use the Ethernet connection instead of the Bluetooth.

The data populates correctly in the SpotData table and when running with -v5 it even displays correctly on the command line.

However, I get no data uploaded to PVOutput for the new inverters.
+I don't see the new inverters data in the vwPvoData table.
+DayData looks like this:
1576266600|2007314220|42437882|252|1   <--Old inverter
1576266600|2007314765|41541572|252|1   <--Old inverter
1576266600|-1288140755|22104|252|          <-New inverter
1576266600|-1288140757|20552|264|          <--New inverter

One of my serial numbers are:
3006826541 which in unsigned hex is B338 882D (notice the high bit is set)
If taken as a signed number that gives -1288140755

Can someone help me locate where the downloaded data gets pumped into sqlite?

The serial number appears correct in the inverter table as well.  I'm sure this is easy to fix I just can't find it on my own.

sbf-, 2019-12-13:
This is fixed in V3.5.1
What version are you running?

bshaffer82, 2019-12-13:
Oh I feel dumb.  I spent all day trying to run down that bug.  

Are there release notes somewhere?  If so maybe a link to them on the wiki.  I saw that the "upload service" was at 3.5.0 so I didn't look any further.

bshaffer82, 2019-12-14:
Also what's the best method of updating.  Should I just move my config files or is there a script?

sbf-, 2019-12-14:
There is a [modification history](https://github.com/SBFspot/SBFspot/wiki/Modification-History) in the wiki (look at the menu at the right side of the wiki page)
The serial number bug is not described as clear as it should however. I'm talking about #269 
The easiest way to update is [sbfspot-config](https://github.com/SBFspot/sbfspot-config)
Before starting, you should backup sbfspot.db and your .cfg files.
Furthermore, using sbfspot-config, you still can't update to 3.6.0 as I'm still working on the MQTT part.

closed 2019-12-21 by sbf-