SBFspot/SBFspot #344: creating config options for changing csv output file for day/month

wasrek404, 2019-12-17:
Hi,

I was trying to get telegraf to read all csv files (not only the name-Spot-20191216.csv). The problem is that telegraf only understands wildcards but no complex regex. So reading a different csv with a similar name but different headers is not possible (I think). Currently following files are created:

$ ls -lah
total 24K
drwxr-xr-x 3 pi pi 4.0K Dec 17 00:42 .
drwxr-xr-x 3 pi pi 4.0K Dec 17 00:42 ..
drwxr-xr-x 2 pi pi 4.0K Dec 17 00:42 Events
-rw-r--r-- 1 pi pi  435 Dec 17 00:56 kas.lan-20191217.csv
-rw-r--r-- 1 pi pi  394 Dec 17 00:56 kas.lan-201912.csv
-rw-r--r-- 1 pi pi 1.3K Dec 17 00:56 kas.lan-Spot-20191216.csv

I use in telegraf these options:

[[inputs.file]]
  files = ["/home/pi/smadata/*/kas.lan-Spot-*.csv"]
  data_format = "csv"
  csv_header_row_count = 1
  csv_delimiter = ";"
  csv_timestamp_column = "dd/MM/yyyy HH:mm:ss"
  csv_timestamp_format = "02/01/2006 15:04:05"

[[inputs.file]]
  files = ["/home/pi/smadata/*/kas.lan-*.csv"]
  data_format = "csv"
  csv_header_row_count = 1
  csv_delimiter = ";"
  csv_timestamp_column = "dd/MM/yyyy HH:mm:ss"
  csv_timestamp_format = "02/01/2006 15:04:05"

[[inputs.file]]
  files = ["/home/pi/smadata/*/kas.lan-*.csv"]
  data_format = "csv"
  csv_header_row_count = 1
  csv_delimiter = ";"
  csv_timestamp_column = "dd/MM/yyyy"
  csv_timestamp_format = "02/01/2006"

Because kas.lan-20191217.csv and kas.lan-201912.csv are very similar and only a wild card is supported but the files have different headers reading the files with telegraf and sending them to influxdb is not possible. It gives an Error in the log when restarting telegraf that says: 

Error in plugin: timestamp column: dd/MM/yyyy HH:mm:ss could not be found

Is it possible to add an option so I can change the output file to "name_day-20191217.csv" and "name_month-201912.csv"?

Thanks in advance!

sbf-, 2019-12-17:
You can change CSVexport.cpp and recompile SBFspot:
`ExportDayDataToCSV()`
`csvpath << FOLDER_SEP << cfg->plantname << "-" << strftime_t("%Y%m%d", date) << ".csv";`
to
`	csvpath << FOLDER_SEP << cfg->plantname << "_day-" << strftime_t("%Y%m%d", date) << ".csv";`

Do the same for `ExportMonthDataToCSV()`

wasrek404, 2019-12-22:
I found a workaround that is easier. One can add tags in telegraf so the data is inserted into influxdb with a tag. Thanks anyways!

closed 2019-12-22 by wasrek404