SBFspot/SBFspot #345: uploaddaemon not working after sbfspot-config install

tomdewilde, 2019-12-19:
system: pi 4 rev 1.2 with debian buster

I've installed SBFspot first using sbfspot-config, then again tried using the manual method (with the latest 3.6.0 version). SBFspot is working but the daemon cannot start, what should I do (linux newbie) ?

`pi@pi-MAIN:~/temp/SBFspot/SBFspotUploadDaemon $ sudo systemctl status SBFspotUpload
● SBFspotUpload.service - SBFspot Upload Daemon
   Loaded: loaded (/usr/local/bin/sbfspot.3/SBFspotUpload.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Thu 2019-12-19 08:22:14 CET; 8s ago
  Process: 5476 ExecStart=/usr/local/bin/sbfspot.3/SBFspotUploadDaemon (code=exited, status=1/FAILURE)
 Main PID: 5476 (code=exited, status=1/FAILURE)

Dec 19 08:22:14 pi-MAIN systemd[1]: Started SBFspot Upload Daemon.
Dec 19 08:22:14 pi-MAIN SBFspotUploadDaemon[5476]: /usr/local/bin/sbfspot.3/SBFspotUploadDaemon: /usr/lib/arm-linux-gn
Dec 19 08:22:14 pi-MAIN systemd[1]: SBFspotUpload.service: Main process exited, code=exited, status=1/FAILURE
Dec 19 08:22:14 pi-MAIN systemd[1]: SBFspotUpload.service: Failed with result 'exit-code'.
`

`pi@pi-MAIN:~/temp/SBFspot/SBFspotUploadDaemon $ /usr/local/bin/sbfspot.3/SBFspotUploadDaemon
/usr/local/bin/sbfspot.3/SBFspotUploadDaemon: /usr/lib/arm-linux-gnueabihf/libcurl.so.4: version `CURL_OPENSSL_3' not found (required by /usr/local/bin/sbfspot.3/SBFspotUploadDaemon`

tomdewilde, 2019-12-19:
seems to be a duplicate of #331 and #295 . But when I try the work around as explained there. I get a permission denied issue. Although I think all permissions should be ok ?

`pi@pi-MAIN:/usr/local/bin/sbfspot.3 $ ./SBFspotUploadDaemon
-bash: ./SBFspotUploadDaemon: Permission denied
pi@pi-MAIN:/usr/local/bin/sbfspot.3 $ ls -all
total 1628
drwxr-xr-x 2 root root   4096 Dec 19 09:39 .
drwxr-xr-x 3 root root   4096 Dec 18 20:37 ..
-rw-r--r-- 1 root root  35855 Dec 19 08:18 date_time_zonespec.csv
-rwxr-xr-x 1 root root    182 Dec 18 21:01 daydata
-rwxr-xr-x 1 root root    191 Dec 18 21:01 monthdata
-rwxr-xr-x 1 root root 330112 Dec 19 08:18 SBFspot
-rw-r--r-- 1 root root   2449 Dec 18 21:01 SBFspot.cfg
-rw-r--r-- 1 root root   9557 Dec 19 08:18 SBFspot.default.cfg
-rw-r--r-- 1 root root   1532 Dec 18 21:01 SBFspotUpload.cfg
-rw-r--r-- 1 root root  70895 Dec 19 09:38 SBFspotUploadDaemon
-rw-r--r-- 1 root root    265 Dec 18 21:01 SBFspotUpload.service
-rw-r--r-- 1 root root 195699 Dec 19 08:18 TagListDE-DE.txt
-rw-r--r-- 1 root root 187840 Dec 19 08:18 TagListEN-US.txt
-rw-r--r-- 1 root root 198143 Dec 19 08:18 TagListES-ES.txt
-rw-r--r-- 1 root root 201627 Dec 19 08:18 TagListFR-FR.txt
-rw-r--r-- 1 root root 196329 Dec 19 08:18 TagListIT-IT.txt
-rw-r--r-- 1 root root 190409 Dec 19 08:18 TagListNL-NL.txt
-rw-r--r-- 1 root root   1142 Dec 19 08:18 Update_340_SQLite.sql
`

sbf-, 2019-12-19:
`sudo chmod +x /usr/local/bin/sbfspot.3/SBFspotUploadDaemon` ?

tomdewilde, 2019-12-19:
I thought I did just that, but just gave it another try and now it works. Thanks again.

closed 2019-12-19 by tomdewilde

stuartcol, 2020-05-22:
Similar Problem

pi@raspberrypi://usr/local/bin/sbfspot.3 $ sudo chmod +x /usr/local/bin/sbfspot.3/SBFspotUploadDaemon
pi@raspberrypi://usr/local/bin/sbfspot.3 $ sudo systemctl start SBFspotUpload.service
pi@raspberrypi://usr/local/bin/sbfspot.3 $ sudo systemctl status SBFspotUpload
● SBFspotUpload.service - SBFspot Upload Daemon
   Loaded: loaded (/usr/local/bin/sbfspot.3/SBFspotUpload.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Fri 2020-05-22 10:18:17 AEST; 5s ago
  Process: 2143 ExecStart=/usr/local/bin/sbfspot.3/SBFspotUploadDaemon (code=exited, status=1/FAILURE)
 Main PID: 2143 (code=exited, status=1/FAILURE)

May 22 10:18:17 raspberrypi systemd[1]: Started SBFspot Upload Daemon.
May 22 10:18:17 raspberrypi SBFspotUploadDaemon[2143]: /usr/local/bin/sbfspot.3/SBFspotUploadDaemon: /usr/lib/arm-linux-gnueabihf/libcurl.so.4: version `CURL
May 22 10:18:17 raspberrypi systemd[1]: SBFspotUpload.service: Main process exited, code=exited, status=1/FAILURE
May 22 10:18:17 raspberrypi systemd[1]: SBFspotUpload.service: Failed with result 'exit-code'.

sbf-, 2020-05-22:
See #295 for workaround