SBFspot/SBFspot #347: SMA time is driving me CRAZY please help

bshaffer82, 2019-12-20:
I've got multiple SMA inverters and I know how SBFspot works.  

1.) The SBFspot always has a problem getting/setting the plant time on my style inverter.  I have 3 of the SB X000US-12 (8000,7000,5000).  Maybe my issue is related to that...not sure.  The 7000 and 5000 seem to work well but perhaps they are just keeping the time I sent with Sunny Explorer.

2.) I pulled logs this morning and as you can see in this file the plant time matches the computer time.
[8k_v5.txt](https://github.com/SBFspot/SBFspot/files/3989935/8k_v5.txt)
[8k_d5.txt](https://github.com/SBFspot/SBFspot/files/3989936/8k_d5.txt)
[8k_settime.txt](https://github.com/SBFspot/SBFspot/files/3989937/8k_settime.txt)

3.) Later my PVoutput data is lagging by ~1 hour.  So I pull a log again.
[8k_v5_later.txt](https://github.com/SBFspot/SBFspot/files/3989949/8k_v5_later.txt)

Much to my surprise, the inverter time has changed.  Not by exactly 1 hour but a different time.  My guess is that earlier in the day (when it wasn't making power) it wasn't really returning time data and PVoutput was using it's own internal lock.  But later once it's making power it actually starts reporting it's real time clock?

sbf-, 2019-12-21:
Setting the time with SBFspot has never been a bullet-proof solution. I still don't understand why as on my inverter it always work.
I propose to disable the timeset function with `SynchTime=0` in the config and set the time once and a while with Sunny Explorer.

sbf-, 2019-12-21:
Maybe you can help by sniffing the USB data when setting the time with Sunny Explorer. USBlyzer is a nice tool to do the job.

bshaffer82, 2019-12-21:
Oh man is capturing bluetooth packets hard.

I thought more info was better than less.  So here is a capture of packets. 
[setting_SMA_time.txt](https://github.com/SBFspot/SBFspot/files/3991585/setting_SMA_time.txt)

Are they are sending unix epoc time which for the time I was trying to set is:
-12/21/2019 @ 11:33AM which I think is 1576927980 or 5DFE02EC.
-But it might be 1576909980 UTC? or 5DFDBC9C
-Problem is none of those actually appear in the file.  :-(
[image]

Searching the above dump file for 5D and looking for FE shows promising results.  But it doesn't match the rest of the hex.  Maybe I'm not doing the conversion to UTC?

I hope these are the packets you need.  Without the SMA protocol translating these packets and finding them amongst a sea of USB packets was hard.  If this isn't the correct one let me know.

sbf-, 2019-12-21:
Makes sense...
```
1b 00 e0 29 43 4c 8f 9b ff ff 00 00 00 00 09 00
00 01 00 09 00 02 03 67 00 00 00 0c 00 63 00 5f
00 70 00 0b ff b5 01 7e 5a 00 24 bd 46 5a 76 02
00 ff ff ff ff ff ff 01 00 
7e
ff 03 60 65
10 a0
ff ff ff ff ff ff	Destination (broadcast)
00 00
78 00 bf 87 3d 39	Source
00 00
00 00
00 00			
09 81				Pkg ID

0a 02 00 f0
00 6d 23 00			Set Time
00 6d 23 00			Set Time
00 6d 23 00			Set Time
81 49 fe 5d			1576946049=Sat, 21 Dec 2019 16:34:09 GMT
81 49 fe 5d			1576946049=Sat, 21 Dec 2019 16:34:09 GMT
81 49 fe 5d			1576946049=Sat, 21 Dec 2019 16:34:09 GMT
b0 b9 ff ff			-18000/3600=UTC-5 (DST off)
0b 00 00 00			???
01 00 00 00			???
da 56				CRC
7e 86  
```
but I can't find the inverter's answer atm

Consider my proposal to disable timesetting by SBFspot to see if it gets better.

bshaffer82, 2019-12-21:
Ok I'll find another way.  The only problem is that the other inverters are located in a different state.  I'll try and get some to go over with a windows laptop.

closed 2020-05-31 by sbf-