SBFspot/SBFspot #349: SBFspot on Intel NUC (Linux Mint/Ubuntu)

Itsmmai, 2019-12-23:
Hello,

Today I bought myself a new Intel NUC i3etc.... Now I want to stop running SBFspot on my pi and start running on my NUC. Is there a manual for this ? I only see arm and pi related stuf.

kind regards,
Ab

anderdd1, 2019-12-23:
I am running it on raspbian raspberry pi vm, so i386 architecture, and all it took was a build/recompile with the correct prerequisite libraries already installed.

Itsmmai, 2019-12-23:
> I am running it on raspbian raspberry pi vm, so i386 architecture, and all it took was a build/recompile with the correct prerequisite libraries already installed.

I really have no clue what you mean and how I should do that ;-) In the pre Debian 10 time it was easy but now I completely lost it :-)

anderdd1, 2019-12-24:
Well, I didn't document it, I just figured it out using my experience.  First thing is you will need a Linux OS (I am using Debian) to run on the machine.  Once you have a functioning Linux OS up then you can use yum to install modules needed, running the make file will indicate missing libraries, and eventually after getting all the modules you need it will compile and then you can following the thorough documentation to manually setup SBFSpot.

The SBFSpot Setup script won't work, you will have to follow the manual setup/install instructions.

closed 2020-02-17 by sbf-

Tokn59, 2024-07-23:
Any progress on this ? Anyone succeeded ?