SBFspot/SBFspot #350: Limitation for MQTT keywords?

rucksman, 2020-01-09:
I installed the MQTT version today and for the sake of testing added all the keywords into the cfg file:
````
MQTT_Data=Timestamp,InvTime,SunRise,SunSet,InvSerial,InvName,InvClass,InvType,InvSwVer,InvStatus,InvTemperature,InvGridRelay,ETotal,EToday,PACTot,PDC1,PDC2,UDC1,UDC2,IDC1,IDC2,OperTm,FeedTm,PAC1,PAC3,UAC1,UAC2,PAC2,UAC3,IAC1,IAC2,IAC3,GridFreq,BatTmpVal,BatVol,BatAmp,BatChaStt
````
But after firing
````
/usr/local/bin/sbfspot.3/SBFspot -finq -nocsv -nosql -v5 -mqtt
````
I only get some values back (partly anonymized with XXXXX):
````
{"Timestamp": "09/01/2020 18:08:45","InvTime": "09/01/2020 18:08:45","SunRise": "XXXXX","SunSet": "XXXXX","InvSerial": XXXXX,"InvName": "SN: XXXXX","InvClass": "Solar-Wechselrichter","InvType": "SB 3600SE-10","InvSwVer": "02.05.02.R","InvStatus": "Ok","InvTemperature": 0.000,"InvGridRelay": "Information liegt nicht vor","ETotal": XXXXX,"EToday": XXXXX,"PACTot": 0.000,"PDC1": 0.000,"PDC2": 0.000,"UDC1": 0.000,"UDC2": 0.000,"IDC1": 0.000,"IDC2": 0.000,"OperTm": XXXXX,"FeedTm": XXXXX,"PAC1": 0.000,"PAC3": 0.000}
````
If I then do only for example
````
MQTT_Data=GridFreq,BatTmpVal
````
I get a result:
````
{"GridFreq": 0.000,"BatTmpVal": 0.000}
````

So is there any limitation in the number of keywords I can use?

sbf-, 2020-01-10:
No, there shouldn't be no limitation, but I need to test is...

rucksman, 2020-01-11:
Would be great, because after experimenting a lot I still experience this behaviour. Of course in production I would not need all keywords, but it would be good to know how many keywords can be used. Thank you!

sbf-, 2020-01-12:
It's a bug in SBFspot.
The inputbuffer in GetConfig() function is too short: only 200 bytes so the MQTT_Data config entry is truncated.
Without rewriting the function, the easiest fix is to increase the buffer:
`    char line[200];
`
to at least 300

poggialb, 2020-06-12:
I have hit a similar issue, not sure if it was the same bug; easy workaround I took was to limit the fields selected until fixed.
In my case the json generated was malformed, causing HA to silently fail parsing, see example below at the end of the payload.

`{"InvTime": "12/06/2020 08:34:30","SunRise": "12/06/2020 07:22:00","SunSet": "12/06/2020 17:08:00","InvSerial": XXX,"InvName": "XXX","InvStatus": "Ok","InvTemperature": 27.330,"InvGridRelay": "Closed","ETotal": XXX,"EToday": XXX,"PACTot": 1625.000,"PDC1": 799.000,"PDC2": 802.000,"UDC1": 350.740,"UDC2": 343.290,"IDC1": 2.281,"IDC2": 2.339,"OperTm": XXX,"FeedTm": XXX,"PAC1": 544.000,"PAC2": 544.000,"PAC3": 544.000,"UAC1": 241.980,"UAC2": 241.980,"UAC3": 241.980,"IAC1": 2.252,"IAC2": 2.252,"IAC3": 2.252,": }`

sbf-, 2020-06-19:
@poggialb Thanks for the well written [hands-on](https://community.home-assistant.io/t/integration-with-sma-solar-stp-10000tl-20-via-sbfspot/204278)

closed 2020-06-19 by sbf-