SBFspot/SBFspot #351: Cannot install SBFspotUploadDaemon

roelvdh, 2020-01-09:
I installed SBFspot via sbfspot-config. It seems to run fine with my 2 inverters (once we get some sunshine here I will know for sure).
The upload-deamon could not be installed, however: /boost/version.hpp doesn't exist. Code:

`pi@raspberrypi:~/SBFspot/SBFspotUploadDaemon $ cd ~/SBFspot/SBFspotUploadDaemon
pi@raspberrypi:~/SBFspot/SBFspotUploadDaemon $ make sqlite
test -d sqlite/obj/ || mkdir -p sqlite/obj/
test -d sqlite/obj//../SBFspotUploadCommon || mkdir -p sqlite/obj//../SBFspotUploadCommon
test -d sqlite/obj//../SBFspot || mkdir -p sqlite/obj//../SBFspot
test -d sqlite/bin/ || mkdir -p sqlite/bin/
g++ ../SBFspotUploadCommon/PVOutput_x.cpp -c -Wall -Wno-unused-local-typedefs -DUSE_SQLITE  -o sqlite/obj/../SBFspotUploadCommon/PVOutput_x.o -I../SBFspot
In file included from ../SBFspotUploadCommon/../SBFspot/SBFspot.h:46:0,
                 from ../SBFspotUploadCommon/PVOutput.h:38,
                 from ../SBFspotUploadCommon/PVOutput_x.cpp:36:
../SBFspotUploadCommon/../SBFspot/boost_ext.h:45:29: fatal error: boost/version.hpp: Bestand of map bestaat niet
 #include <boost/version.hpp>
                             ^
compilation terminated.
makefile:106: recept voor doel 'sqlite/obj/../SBFspotUploadCommon/PVOutput_x.o' is mislukt
make: *** [sqlite/obj/../SBFspotUploadCommon/PVOutput_x.o] Fout 1
pi@raspberrypi:~/SBFspot/SBFspotUploadDaemon $
`

Any ideas?

roelvdh, 2020-01-10:
FYI: the Pi is a PI-Zero

sbf-, 2020-01-10:
SBFspotUploadDaemon should be installed with sbfspot-config
You try to build it without the necessary BOOST libraries installed. See [wiki](https://github.com/SBFspot/SBFspot/wiki/Installation-Linux-SQLite)

roelvdh, 2020-01-11:
Thank you, after installing Boost it worked like a charm. May I please add two remarks, intended to help you improving an already excellent wiki.

1. Add a line sudo apt-get update before the sbfspot-config line
My Pi-Zero came with a ready-made NOOBS SD-card. Only later, I found out the version of Raspbian was very, very old. This must have caused sbfspot-config not to install SBFSpotUpdateDeamon, but I haven't noticed that. A Raspbian update before entering into sbfspot-config is a natural thing to do but it might be wise to mention it anyway.

2. Make clear that sbfspot-config installs all necesary programs
As trhe format of the wiki is now, it starts with sbfspot-config, Installation, Database Setup, Configuration, Testing, Automation, and then the same for SBFSpotUpdateDeamon. This could easily suggest - like it did with me - that two installations are necessary and sbfspot-config only includes the first one.

Wishing you all the best with further developments of this great project,

Roel

closed 2020-01-15 by roelvdh

geert-hendrickx-be, 2020-02-11:
boost-filesystem and boost-regex are linked but not used?  are they actually needed?

sbf-, 2020-02-11:
> boost-filesystem and boost-regex are linked but not used? are they actually needed?

hmm... might be a contamination of SBFspot 4 development