SBFspot/SBFspot #353: Feature Request: Support for InfluxDB

philipptrenz, 2020-01-12:
> InfluxDB is an open-source time series database (TSDB) developed by InfluxData. It is written in Go and optimized for fast, high-availability storage and retrieval of time series data in fields such as operations monitoring, application metrics, Internet of Things sensor data, and real-time analytics. – [Wikipedia](https://en.wikipedia.org/wiki/InfluxDB)

TSDBs like [InfluxDB](https://www.influxdata.com) are perfect for the type of data SBFspot produces and bring major advantages. The data gets stores very efficiently, aggregation functions like count, sum, average are integrated within the database and data can also be compressed (something like "for data older than 1 week only keep the average over each hour").

Data from InfluxDB can also easily be visualized cloud-free with [Grafana](https://grafana.com).

The API for InfluxDB is also pretty easy as it uses HTTP requests with its [line protocol](https://docs.influxdata.com/influxdb/v1.7/write_protocols/line_protocol_tutorial/):
```
weather,location=us-midwest temperature=82 1465839830100400200
[|measurement|,tag_set| |field_set| |timestamp|]
```

Providing InfluxDB as a database option would highly reduce overhead and storage space, which is especially interesting for low power devices like the Raspberry Pi. Also with that the data is easily accessible from several home automation systems, like NodeRED.

ced2git, 2021-05-10:
Hello, 

until sbfspot supports natively, there is a python script called: sbfspot2influxdb that does exactly what you are looking for. I've made an enfancement to support v2 of influxdb but still not published.

ZonPV, 2022-03-06:
> Hello,
> 
> until sbfspot supports natively, there is a python script called: sbfspot2influxdb that does exactly what you are looking for. I've made an enfancement to support v2 of influxdb but still not published.

With my knowledge no luck to configure and install that script. So looking forward to the solution of SBFspot.

janknieling, 2022-12-21:
Any update here? Is there an ETA when we can expect to get the InfluxDB support implemented?