SBFspot/SBFspot #354: CRITICAL: Failed to initialize Speedwire

semiceau, 2020-01-16:
i'm recieving this - it is detecting the inverter and the creds are correct.

root@ubuntu:/usr/local/bin/sbfspot.3# /usr/local/bin/sbfspot.3/SBFspot -v5 -finq -nocsv
SBFspot V3.6.0
Yet another tool to read power production of SMA solar inverters
(c) 2012-2019, SBF (https://github.com/SBFspot/SBFspot)
Compiled for Linux (LE) 64 bit with SQLite support
Commandline Args: -v5 -finq -nocsv
Reading config '/usr/local/bin/sbfspot.3/SBFspot.cfg'
Configuration settings:
IP_Address=0.0.0.0
Password=<undisclosed>
MIS_Enabled=0
Plantname=thebrown
OutputPath=/home/pi/smadata/%Y
OutputPathEvents=/home/pi/smadata/%Y/Events
Latitude=50.8
Longitude=4.33
Timezone=Australia/Melbourne
CalculateMissingSpotValues=1
DateTimeFormat=%d/%m/%Y %H:%M:%S
DateFormat=%d/%m/%Y
TimeFormat=%H:%M:%S
SynchTime=1
SynchTimeLow=1
SynchTimeHigh=3600
SunRSOffset=900
DecimalPoint=comma
CSV_Delimiter=semicolon
Precision=3
CSV_Export=0
CSV_ExtendedHeader=1
CSV_Header=1
CSV_SaveZeroPower=1
CSV_Spot_TimeSource=0
CSV_Spot_WebboxHeader=0
Locale=en-US
BTConnectRetries=10
SQL_Database=/home/pi/smadata/SBFspot.db
### End of Config ###
Thu Jan 16 20:42:12 2020: INFO: Starting...
sunrise: 18:39
sunset : 03:05
Connecting to Local Network...
Initializing...
SUSyID: 125 - SessionID: 812500903 (0x306DC7A7)
Inverter IP address: 192.168.69.102 found via broadcastidentification
ERROR: Connection to inverter failed!
Is 192.168.69.102 the correct IP?
Please check IP_Address in SBFspot.cfg!
Thu Jan 16 20:42:17 2020: CRITICAL: Failed to initialize Speedwire connection.]

any ideas?

sbf-, 2020-01-18:
Related to #256 
Do you have sma home manager or energy meter in your network?

semiceau, 2020-01-18:
no just the inverter :|

lawroman, 2020-03-18:
I had a similar problem too, it happened once in a while.  My wireless connection on the Linux box was unstable, I switched it to wire and have not seen the problem.

sbf-, 2020-04-06:
Is the inverter wireless connected?

Not related to this issue: you should change lat/lon to your location in order to have a correct sunrise/sunset time
```
Latitude=50.8  # wrong!
Longitude=4.33 # wrong!
Timezone=Australia/Melbourne

sunrise: 18:39
sunset : 03:05
```

closed 2020-05-31 by sbf-