SBFspot/SBFspot #355: running on debian (vsphere) not working

sheevz0r, 2020-01-17:
hello there,

i'm trying to get this working on a debian server,
Installing went fin till this:
 `root@sunnyboy:~/SBFspot/SBFspot# ./SBFspot.h -v5 -mqtt => giving acces denied`

i chmodded 755 on the whole folder,and now getting this:
```
root@sunnyboy:~/SBFspot/SBFspot# ./SBFspot.h -v5 -mqtt
./SBFspot.h: line 1: /**************************************************************************************: No such file or directory
./SBFspot.h: line 2: SBFspot: command not found
./SBFspot.h: line 3: syntax error near unexpected token `2012-2019,'
'/SBFspot.h: line 3: `  (c)2012-2019, SBF
```
i followed your guide step by step.

sbf-, 2020-01-18:
> i followed your guide step by step.

Not so sure about that 😉

The command is
`/usr/local/bin/sbfspot.3/SBFspot -v5 -mqtt`

sheevz0r, 2020-01-18:
https://github.com/SBFspot/SBFspot/wiki/MQTT i followed this :/ 
```
root@sunnyboy:~/SBFspot/SBFspot# /usr/local/bin/sbfspot.3/SBFspot -v5 -mqtt
-bash: /usr/local/bin/sbfspot.3/SBFspot: No such file or directory
```

sbf-, 2020-01-18:
Ok I see, there is a missing step...
`sudo make install_sqlite`

sbf-, 2020-01-20:
Wiki updated

closed 2020-01-20 by sbf-