SBFspot/SBFspot #357: Reading of SMA inverter via SMA COM Gateway (Speedwire) does not work.

sonne4711, 2020-02-03:
I have my Inverters (SB5000TL20) connected via RS485 towards a SMA COM Gateway which translates the RS485 into Speedwire (SMA Data 2+). If I connect my SMA Cluster Controller or Sunny Explorer it all works fine.

SBFspot cant recognize all inverter (just finds the first one out of 6) nor can read the values correctly.

I think its a very valid case since there are many SMA/RS485 based setups out there where the SMA Webbox is swapped with the SMA COM Gateway (with cluster Controller or Data Manager).

here is my SBFspot log:

SBFspot V3.5.2
Yet another tool to read power production of SMA solar inverters
(c) 2012-2018, SBF (https://github.com/SBFspot/SBFspot)
Compiled for Linux (LE) 32 bit with SQLite support
Commandline Args: -v -ad1 -am0 -ae0 -cfg/usr/local/bin/sbfspot.3/SBFspot.cfg
Reading config '/usr/local/bin/sbfspot.3/SBFspot.cfg'
Mon Feb  3 16:20:01 2020: INFO: Starting...
sunrise: 07:46
sunset : 16:53
Connecting to Local Network...
Initializing...
SUSyID: 125 - SessionID: 847890935 (0x3289C9F7)
Inverter IP address: 172.22.0.2 from SBFspot.cfg
Logon OK
getSoftwareVersion returned an error: -1
getTypeLabel returned an error: -1
getDeviceStatus returned an error: -1
getInverterTemperature returned an error: -1
getMaxACPower returned an error: -1
getEnergyProduction returned an error: -1
getOperationTime returned an error: -1
getSpotDCPower returned an error: -1
getSpotDCVoltage returned an error: -1
SUSyID: 355 - SN: 2100121xxx
DC Spot Data:
	String 1 Pdc:   0.000kW - Udc:   0.00V - Idc:  0.000A
	String 2 Pdc:   0.000kW - Udc:   0.00V - Idc:  0.000A
getSpotACPower returned an error: -1
getSpotACVoltage returned an error: -1
getSpotACTotalPower returned an error: -1
SUSyID: 355 - SN: 2100121xxx
AC Spot Data:
	Phase 1 Pac :   0.000kW - Uac:   0.00V - Iac:  0.000A
	Phase 2 Pac :   0.000kW - Uac:   0.00V - Iac:  0.000A
	Phase 3 Pac :   0.000kW - Uac:   0.00V - Iac:  0.000A
	Total Pac   :   0.000kW
getSpotGridFrequency returned an error: -1
********************
* ArchiveDayData() *
********************
startTime = 5E375470 -> 03/02/2020 00:00:00
ExportDayDataToCSV()

I hope we can address this Bug/Feature nicely.
--thomas

sbf-, 2020-02-03:
Hmm... “logon ok” is a good start
Can you post the output of SBFspot -d5 -v5

sonne4711, 2020-02-04:
Yeah I was happy that the connection is working at least. In the response I see that it can pull the daily 5 minutes values. But the inverter discovery and "live" values cannot be red.... please have a look at the attached log files.

[sbfspot_output_debug.zip](https://github.com/SBFspot/SBFspot/files/4152164/sbfspot_output_debug.zip)

--thomas

sbf-, 2020-02-04:
It seems you get responses from all 6 inverters and the cluster controller (or com gateway - not sure about that), but SBFspot doesn't handle the data. Even worse, it expects a response to another request and doesn't know what to do with the login reply of the inverters.
I'll see what I can do, but I'm afraid it will take some effort. I have to be sure not to disrupt the current login procedure for all the other installations.
FYI: the devices found are (S/N ending with):
```
355:xxxxxxx083
355:xxxxxxx715
355:xxxxxxx716
355:xxxxxxx271
355:xxxxxxx260
355:xxxxxxx372
354:xxxxxxx054 (Cluster Controller?)
```

sonne4711, 2020-02-04:
Thanks for the quick feedback!

Indeed the serial numbers are correct and in the same order as shown in the sunny explorer or cluster controller. The last serial number  (054) is of the com gateway itself.

So the com gateway broadcasts automatically all data after login? 

I'm going to the physical location of the system and would be able to do some specific investigations if needed. I were thinking of connecting the Sunny Explorer and to record the network traffic with an sniffer if it would be of help.

--thomas

sbf-, 2021-03-26:
@sonne4711 Is this still relevant/needed? If yes, a network trace to SE would be nice.

sonne4711, 2021-04-19:
Actually, it would be still good to have this feature. Sorry, that I did
not answer realier but i did not recognize the last email.
What exactly you mean by  "a network trace to SE". Maybe I can support you
with it.

kr,

--thomas

sbf-, 2021-04-19:
Ok, no problem, I was cleaning up some older issues...
Well, like you said before:
>I were thinking of connecting the Sunny Explorer and to record the network traffic with an sniffer if it would be of help.

closed 2024-06-18 by sbf-