SBFspot/SBFspot #360: Archive Data into MariaDB/MySQL

PaulDe20, 2020-02-10:
Hey,
I've got sbfspot working on my raspberry Pi3+ for a SunnyBoy 5000.

Ist it possible to get all the archived Data from the converter into the Database?
Right now sbfspot logs the Daydata and in the end of the day it writes it into monthdata.
I think the converter would have more information on the internal storage?

Thanks!

saylinglow, 2020-02-10:
You can run SBFspot from the command line to collect the previous data you want:
bash:   /usr/local/bin/sbfspot.3/SBFspot -v -ad60 -am2 -finq -nocsv 
would get you the past 60 days/2 months of data from the inverter to the SQL database. (-finq would force a connection to the inverter if the time is after sunset/before sunrise)
bash:  /usr/local/bin/sbfspot.3/SBFspot -?
will list the help text and give you the switches for the program. 

My guess is you used sbfspot-config to install the program. There is a cron job which runs SBFspot every 5 minutes between 0600 and 2200 to interrogate the inverter and populate the SQL database. If you are uploading the data to PVOutput, SBFspotUpload runs as a linux service to upload new entries in the SQL database up to the PVOutput server.

Sayling

PaulDe20, 2020-02-10:
Hey,
thanks alot for this fast answer!

Great! :)

Paul

PaulDe20, 2020-02-10:
So,
after this works fine, I've found another problem for me, while trying to generate some statistics.
I got the MonthData for the last 9 years.
But, always in October I have "32 days" logged and in March I miss one. I think it is cause of the time change.
I read this from MySQL by:
**`select FROM_UNIXTIME(TimeStamp) as Datum from MonthData order by Datum`**

Is it possible to correct this on an easy way? In October there is an entry for the "double day" at 23:00 and 00:00

Thanks!

sbf-, 2020-02-11:
First, execute select query to verify:
```SQL
select *, FROM_UNIXTIME(TimeStamp) as Datum
from MonthData
where FROM_UNIXTIME(TimeStamp)='2010-12-21 00:00:00'
```
When OK, delete one of the two records:
```SQL
delete from MonthData
where FROM_UNIXTIME(TimeStamp)='2010-12-21 00:00:00'
```

sbf-, 2020-02-11:
To find double dates, try this:
```SQL
SELECT 
    DATE(FROM_UNIXTIME(TimeStamp)) As Datum, 
    COUNT(DATE(FROM_UNIXTIME(TimeStamp))) As CountDatum
FROM
    monthdata
GROUP BY DATE(FROM_UNIXTIME(TimeStamp))
HAVING COUNT(DATE(FROM_UNIXTIME(TimeStamp))) > 1;
```

closed 2020-02-17 by sbf-