SBFspot/SBFspot #362: SBFspot archive data to mariadb Out of range value for column 'Serial'

deagel, 2020-02-16:
Hi all,
first SBFspot is a very nice tool thanks for this.

View background information’s:
i run long time sbfspot 3.5 on my old pi3 with sqlight, now i got an pi4 and like to use an "real" DB -> MariaDB. The migration run OK but i think there are still view thinks in the DB schema itself. As i started with sbfspot there are a view workaround to rebuild the dayly view with out of range warnings. 
In the migration process i used a insert ignore command to brin my data in the new db, i now that there was a lot of entry’s that was not imported. so i try to get the archive date from my inverter directly.

/usr/local/bin/sbfspot.3/SBFspot -v -ad300 -am300 -ae300 -cfg/usr/local/bin/sbfspot.3/SBFspot.cfg

I see a lot of messages like:

startTime = 5E126A70 -> 06/01/2020 00:00:00
ExportDayDataToCSV()
[2020-02-16 08:40:30.635] Error: [day_data]mysql_stmt_execute() returned : Out of range value for column 'Serial' at row 1

and

********************
* ArchiveDayData() *
********************
startTime = 5E278270 -> 22/01/2020 00:00:00
[2020-02-16 09:00:11.486] Error: [day_data]mysql_stmt_execute() returned : Out of range value for column 'Serial' at row 1

Here view information’s about my inverters:

Device Name:      SB 2.5 Fassade
Device Class:     Solar-Wechselrichter
Device Type:      SB2.5-1VL-40
Software Version: 02.05.01.R
Serial number:    1000000003
SUSyID: 361 - SN: 3000000009

Device Name:      SBS 3.7 Li
Device Class:     Batterie-Wechselrichter
Device Type:      Sunny Boy Storage 3.7
Software Version: 03.00.30.R
Serial number:    3000000009
SUSyID: 235 - SN: 2000000004

Device Name:      SB 2500TLST-21 Süd Dach
Device Class:     Solar-Wechselrichter
Device Type:      SB 2500TLST-21
Software Version: 02.80.22.R
Serial number:    2000000004
SUSyID: 361 - SN: 3000000009

Thanks,
Dennis

sbf-, 2020-02-17:
I think this is related to #330

deagel, 2020-02-18:
Duplicate of #330

closed 2020-02-18 by deagel