SBFspot/SBFspot #363: SBFspotUploadDaemon CURL_OPENSSL_3 not found

deagel, 2020-02-16:
Hi,
i installed yesterday with sbfspot-config. Today i checked my uploads but there are no uploads.

hi have more or less a min. install with NOOBS_v3_3_0 is there an missing dep. in the configure skript?

root@raspberrypi-4:/var/log# systemctl status SBFspotUpload.service
* SBFspotUpload.service - SBFspot Upload Daemon
   Loaded: loaded (/usr/local/bin/sbfspot.3/SBFspotUpload.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Sun 2020-02-16 10:34:45 CET; 4min 0s ago
  Process: 13609 ExecStart=/usr/local/bin/sbfspot.3/SBFspotUploadDaemon (code=exited, status=1/FAILURE)
 Main PID: 13609 (code=exited, status=1/FAILURE)

Feb 16 10:34:45 raspberrypi-4 systemd[1]: Started SBFspot Upload Daemon.
Feb 16 10:34:45 raspberrypi-4 SBFspotUploadDaemon[13609]: /usr/local/bin/sbfspot.3/SBFspotUploadDaemon: /usr/lib/arm-linux-gnueabihf/libcurl.so.4: version `CURL_OPENSSL_3' not found (required by /usr/local/bin
Feb 16 10:34:45 raspberrypi-4 systemd[1]: SBFspotUpload.service: Main process exited, code=exited, status=1/FAILURE
Feb 16 10:34:45 raspberrypi-4 systemd[1]: SBFspotUpload.service: Failed with result 'exit-code'.

saylinglow, 2020-02-16:
This looks to me like closed issue #295 (and #331) https://github.com/SBFspot/SBFspot/issues/295https://github.com/SBFspot/SBFspot/issues/295
Sayling

deagel, 2020-02-16:
Hi, yes look like #295 and #331 .
nice to have the workaround in a final solution that the bfspot-config handel it. ;)

closed 2020-02-16 by deagel