SBFspot/SBFspot #364: CRITICAL: Logon failed. Check 'USER' Password

yustmenl, 2020-02-21:
I am trying to install SBF spot for a SB5000 and i tryed to connect over BT an LAN with both i get the error CRITICAL: Logon failed. Check 'USER' Password

when on BT is see in the invertor log 
"Parameter "Set user password" was sent (10222) "

i tryed to change the PWD and then use that for connecting but that also didn't work

Hopefully anyone has an idea/solution

sbf-, 2020-02-24:
Please post debuglog -d5 -v5

rucksman, 2020-02-24:
Maybe this is not a solution in general, but from time to time I had the same issue. Since I call the cronjob with "-installer -password:%hereGoesYourInstallerPassword%" the login is very reliable.

yustmenl, 2020-02-27:
here is the degug log

SBFspot V3.5.2
Yet another tool to read power production of SMA solar inverters
(c) 2012-2018, SBF (https://github.com/SBFspot/SBFspot)
Compiled for Linux (LE) 32 bit with SQLite support
Commandline Args: -v -nocsv -nosql -finq -password:******* -d5 -v5
Reading config '/usr/local/bin/sbfspot.3/SBFspot.cfg'
Configuration settings:
BTAddress=00:80:25:33:95:25
Password=<undisclosed>
MIS_Enabled=0
Plantname=yustmenl
OutputPath=/home/pi/smadata/%Y
OutputPathEvents=/home/pi/smadata/%Y/Events
Latitude=51.51
Longitude=-0.13
Timezone=Europe/Brussels
CalculateMissingSpotValues=0
DateTimeFormat=%d/%m/%Y %H:%M:%S
DateFormat=%d/%m/%Y
TimeFormat=%H:%M:%S
SynchTime=1
SynchTimeLow=1
SynchTimeHigh=3600
SunRSOffset=900
DecimalPoint=comma
CSV_Delimiter=semicolon
Precision=3
CSV_Export=0
CSV_ExtendedHeader=1
CSV_Header=1
CSV_SaveZeroPower=1
CSV_Spot_TimeSource=0
CSV_Spot_WebboxHeader=0
Locale=en-US
BTConnectRetries=10
SQL_Database=/home/pi/smadata/SBFspot.db
### End of Config ###
Thu Feb 27 07:17:37 2020: INFO: Starting...
sunrise: 06:51
sunset : 17:34
Connecting to 00:80:25:33:95:25 (1/10)
Initializing...
SUSyID: 125 - SessionID: 928968168 (0x375EEDE8)
getPacket(2)
MAX_CommBuf is now 18 bytes
Received 18 bytes
Received 13 bytes
--------: 00 01 02 03 04 05 06 07 08 09
00000000: 7E 1F 00 61 25 95 33 25 80 00
00000010: 00 00 00 00 00 00 02 00 00 04
00000020: 70 00 06 00 00 00 00 01 00 00
00000030: 00
cmd=2
<<<====== Content of pcktBuf =======>>>
--------: 00 01 02 03 04 05 06 07 08 09
00000000: 7E 1F 00 61 25 95 33 25 80 00
00000010: 00 00 00 00 00 00 02 00 00 04
00000020: 70 00 06 00 00 00 00 01 00 00
00000030: 00
<<<=================================>>>
MAX_pcktBuf is now 31 bytes
SMA netID=06
--------: 00 01 02 03 04 05 06 07 08 09
00000000: 7E 1F 00 61 00 00 00 00 00 00
00000010: 25 95 33 25 80 00 02 00 00 04
00000020: 70 00 06 00 00 00 00 01 00 00
00000030: 00
31 Bytes sent
[... 114 lines cut ...]
Received 18 bytes
Received 67 bytes
--------: 00 01 02 03 04 05 06 07 08 09
00000000: 7E 55 00 2B 25 95 33 25 80 00
00000010: 3A 82 D8 EB 27 B8 01 00 7E FF
00000020: 03 60 65 0E D0 7D 5D 00 E8 ED
00000030: 5E 37 00 01 8A 00 A2 8A FB 7D
00000040: 5E 00 01 00 01 00 00 04 80 0D
00000050: 04 FD FF 07 00 00 00 84 03 00
00000060: 00 7D 31 6D 57 5E 00 00 00 00
00000070: B8 B8 B8 B8 88 88 88 88 88 88
00000080: 88 88 5E 43 7E
cmd=1
PacketLength=85
<<<====== Content of pcktBuf =======>>>
--------: 00 01 02 03 04 05 06 07 08 09
00000000: 7E FF 03 60 65 0E D0 7D 00 E8
00000010: ED 5E 37 00 01 8A 00 A2 8A FB
00000020: 7E 00 01 00 01 00 00 04 80 0D
00000030: 04 FD FF 07 00 00 00 84 03 00
00000040: 00 11 6D 57 5E 00 00 00 00 B8
00000050: B8 B8 B8 88 88 88 88 88 88 88
00000060: 88 5E 43 7E
<<<=================================>>>
Thu Feb 27 07:17:37 2020: CRITICAL: Logon failed. Check 'USER' Password

sbf-, 2020-02-27:
You pass the password via command line, but this is only in combination with `-installer`
For "user" the password should be in the config file (now login password is by default 0000 which is probably incorrect as you have modified it)

yustmenl, 2020-03-03:
> You pass the password via command line, but this is only in combination with `-installer`
> For "user" the password should be in the config file (now login password is by default 0000 which is probably incorrect as you have modified it)

Correct, i now passed the password with the commandline this time.
do you want the log without the pass in the command line?

yustmenl, 2020-03-03:
here is the result without the pass from the commandline

/usr/local/bin/sbfspot.3/SBFspot -v -nocsv -nosql -finq  -d5 -v5
SBFspot V3.5.2
Yet another tool to read power production of SMA solar inverters
(c) 2012-2018, SBF (https://github.com/SBFspot/SBFspot)
Compiled for Linux (LE) 32 bit with SQLite support
Commandline Args: -v -nocsv -nosql -finq -d5 -v5
Reading config '/usr/local/bin/sbfspot.3/SBFspot.cfg'
Configuration settings:
BTAddress=00:80:25:33:95:25
Password=<undisclosed>
MIS_Enabled=0
Plantname=yustmenl
OutputPath=/home/pi/smadata/%Y
OutputPathEvents=/home/pi/smadata/%Y/Events
Latitude=51.51
Longitude=-0.13
Timezone=Europe/Brussels
CalculateMissingSpotValues=0
DateTimeFormat=%d/%m/%Y %H:%M:%S
DateFormat=%d/%m/%Y
TimeFormat=%H:%M:%S
SynchTime=1
SynchTimeLow=1
SynchTimeHigh=3600
SunRSOffset=900
DecimalPoint=comma
CSV_Delimiter=semicolon
Precision=3
CSV_Export=0
CSV_ExtendedHeader=1
CSV_Header=1
CSV_SaveZeroPower=1
CSV_Spot_TimeSource=0
CSV_Spot_WebboxHeader=0
Locale=en-US
BTConnectRetries=10
SQL_Database=/home/pi/smadata/SBFspot.db
### End of Config ###
Tue Mar  3 18:44:26 2020: INFO: Starting...
sunrise: 06:41
sunset : 17:43
Connecting to 00:80:25:33:95:25 (1/10)
Initializing...
SUSyID: 125 - SessionID: 926750127 (0x373D15AF)
getPacket(2)
MAX_CommBuf is now 18 bytes
Received 18 bytes
Received 13 bytes
--------: 00 01 02 03 04 05 06 07 08 09
00000000: 7E 1F 00 61 25 95 33 25 80 00
00000010: 00 00 00 00 00 00 02 00 00 04
00000020: 70 00 06 00 00 00 00 01 00 00
00000030: 00
cmd=2
<<<====== Content of pcktBuf =======>>>
--------: 00 01 02 03 04 05 06 07 08 09
00000000: 7E 1F 00 61 25 95 33 25 80 00
00000010: 00 00 00 00 00 00 02 00 00 04
00000020: 70 00 06 00 00 00 00 01 00 00
00000030: 00
<<<=================================>>>
MAX_pcktBuf is now 31 bytes
SMA netID=06
--------: 00 01 02 03 04 05 06 07 08 09
00000000: 7E 1F 00 61 00 00 00 00 00 00
00000010: 25 95 33 25 80 00 02 00 00 04
00000020: 70 00 06 00 00 00 00 01 00 00
00000030: 00
[... 115 lines cut ...]
Received 18 bytes
Received 66 bytes
--------: 00 01 02 03 04 05 06 07 08 09
00000000: 7E 54 00 2A 25 95 33 25 80 00
00000010: 3A 82 D8 EB 27 B8 01 00 7E FF
00000020: 03 60 65 0E D0 7D 5D 00 AF 15
00000030: 3D 37 00 01 8A 00 A2 8A FB 7D
00000040: 5E 00 01 00 01 00 00 04 80 0D
00000050: 04 FD FF 07 00 00 00 84 03 00
00000060: 00 8B A5 5E 5E 00 00 00 00 B8
00000070: B8 B8 B8 88 88 88 88 88 88 88
00000080: 88 D5 54 7E
cmd=1
PacketLength=84
<<<====== Content of pcktBuf =======>>>
--------: 00 01 02 03 04 05 06 07 08 09
00000000: 7E FF 03 60 65 0E D0 7D 00 AF
00000010: 15 3D 37 00 01 8A 00 A2 8A FB
00000020: 7E 00 01 00 01 00 00 04 80 0D
00000030: 04 FD FF 07 00 00 00 84 03 00
00000040: 00 8B A5 5E 5E 00 00 00 00 B8
00000050: B8 B8 B8 88 88 88 88 88 88 88
00000060: 88 D5 54 7E
<<<=================================>>>
Tue Mar  3 18:44:27 2020: CRITICAL: Logon failed. Check 'USER' Password

sbf-, 2020-03-04:
The config file must contain the password=your_new_password
Now it looks like it is blank or 0000 or simply missing

yustmenl, 2020-03-06:
Thanks looks like that worked

closed 2020-03-06 by yustmenl