SBFspot/SBFspot #365: No longer uploading to PVoutput

TorpedoBE, 2020-02-22:
Hi All,

Suddenly my SBFspot stopped uploading to PVoutput.org.

In the syslog I see below issue, any idea what's wrong?

SBFspotUploadDaemon: Unable to open output file /tmp/SBFspotUploadDaemon.out: Success

regards,

Bart

sbf-, 2020-02-22:
Reboot maybe?

TorpedoBE, 2020-02-22:
been there, done that without result.....

mw446391, 2020-02-22:
Not sure, if this related, but I had noticed on WED evening that PVoutput showed 0 kWh for the whole day; I rebooted my raspi and after a while the generation data for WED showed up.

Also earlier today (maybe an hour ago or so), I noticed that the generation data had flatlined even though it's quite sunny. But without doing anything on my raspi the data eventually showed up.

TorpedoBE, 2020-02-22:
mine stopped uploading on February 15th without any reason...

If I try /usr/local/bin/sbfspot.3/SBFspot -v -finq -nocsv everthing works fine

in db is everthing stord only uploading to pvoutput doesn't work anymore...

mw446391, 2020-02-22:
I'm running SBFspot V3.5.2 on the latest version of Raspbian GNU/Linux 9 (stretch) on my Raspberry Pi 3 Model B Rev 1.2 and it works.

sbf-, 2020-02-22:
V352 uses another upload mechanism

TorpedoBE, 2020-02-22:
Can I install it on Raspberry pi? in seperate directory?? any manual?

mw446391, 2020-02-22:
@TorpedoBE what version of SBF are you running? 

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

TorpedoBE, 2020-02-23:
I'm running Version 3.1.2.

TorpedoBE, 2020-02-23:
the problem is that I don't execute the daemon as root....

so if I run 

sudo /usr/local/bin/sbfspot.3/SBFspotUploadDaemon

there is no error in the syslog

but still no upload...

TorpedoBE, 2020-02-24:
I decided to install another pi with the new version so this issue can be closed.

closed 2020-02-24 by TorpedoBE