SBFspot/SBFspot #366: SBFspot only works for user pi

HeinvdW, 2020-02-22:
Hi,

I deleted user pi. However being another user SBPsot can be installed, but not be executed.
It tries to connect through bluetooth but connection is refused by the inverter.
There is a problem with ownershop and group.
I am just a newcomer in Linux so I cannot find a way out other than installing SBFspot under user pi.

Could you change the setup of SBFspot that it can be installed and run under other users as well.

sbf-, 2020-02-22:
If you are a linux newcomer, why deleting user pi???

HeinvdW, 2020-02-22:
Security, no name pi present. [https://thepihut.com/blogs/raspberry-pi-tutorials/how-to-change-the-default-account-username-and-password](url)

HeinvdW, 2020-02-22:
Still, is it correct that SBFspot can only be installed for user pi? 
Or did I miss a setting in the instructions.
[https://github.com/SBFspot/SBFspot/wiki/Installation-Linux-SQLite](url)

sbf-, 2020-02-23:
No that’s not correct

99technetium, 2020-02-23:
I'm running SBFspot on a Debian server and therefore not as the user "pi". I only had to change the directory paths in the default configuration file of SBFspot from "/home/pi/..." to "/home/my_user/..."

HeinvdW, 2020-03-02:
I tried several times on Pi, also changed all home/pi for home/myuser. Still everything worked, just the BT connection would not connect. Maybe I should have changed permissions for BT user from pi to myuser. Didn't find that in the install instructions.

sbf-, 2020-03-03:
Show us some output/error messages.
Almost sure it has nothing to do with other username

closed 2020-03-23 by sbf-

HeinvdW, 2020-04-28:
Hi SBF,
You are right. SBFspot can be installed for any user.
I must have overseen something somewhere.
Deleted 3.6.0 and installed it again, carefully following the instructions.
Success!