SBFspot/SBFspot #368: SBFUploadDaemon does not do the upload job with Buster

berbergh, 2020-02-25:
Seems the problem is not yet gone.
running SBFspot 3.5.2 with SQLite on Raspberry Pi (Raspian buster) 
SBFspot fills the SQlite database, but nothing is uploaded to PVoutput.

Logging does this:

On 22 feb 2020:
[17:28:17] INFO: Uploading 30 datapoints, starting with 20200222,10:40,19256834,48,,,24.5,225.61 => OK (200)
[17:30:19] INFO: Uploading 30 datapoints, starting with 20200222,13:10,19257278,276,,,27.5,227.36 => OK (200)
[17:32:20] INFO: Uploading 15 datapoints, starting with 20200222,15:40,19257813,156,,,26.8,228.99 => OK (200)
[19:01:32] WARNING: xxxxx is not yet member of SBFspot Team. Consider joining at http://pvoutput.org/listteam.jsp?tid=613

On 23 feb 2020:
[17:57:18] INFO: Stopping Daemon...
[17:57:39] INFO: SBFspotUploadDaemon Version 2.0.0
[17:57:39] INFO: Starting Daemon...
[19:14:31] WARNING: xxxxxx is not yet member of SBFspot Team. Consider joining at http://pvoutput.org/listteam.jsp?tid=613

Same on 24 feb 2020.

sbf-, 2020-02-25:
Are you sure the data is ok?

saylinglow, 2020-02-25:
Just wanted to say I am running on Raspbian Buster (I built it in Jan 2020) and SBFspot 3.5.2 on SQLite  also and the upload is working fine. I have put my smadata directory (database, logs & CSV data) on a USB memory stick. 

berbergh, does your syslog show any other messages from SBFspotUploadDaemon?
try 
> cat /var/log/syslog | grep SBFspot

berbergh, 2020-02-26:
@saylinglow, this command cat /var/log/syslog | grep SBFspot does not produce anything.

cat /var/log/syslog | grep sbfspot produces this kind of data:
Feb 26 07:45:01 raspberrypi CRON[21225]: (pi) CMD (/usr/local/bin/sbfspot.3/daydata)
Feb 26 07:50:01 raspberrypi CRON[21258]: (pi) CMD (/usr/local/bin/sbfspot.3/daydata)
Feb 26 07:55:01 raspberrypi CRON[21293]: (pi) CMD (/usr/local/bin/sbfspot.3/daydata)

berbergh, 2020-02-26:
@SBF , I am not sure what you mean.

more SMABA1_20200225.log contains data like:
Yet another tool to read power production of SMA solar inverters
(c) 2012-2018, SBF (https://github.com/SBFspot/SBFspot)
Compiled for Linux (LE) 32 bit with SQLite support
Commandline Args: -v -ad1 -am0 -ae0 -cfg/usr/local/bin/sbfspot.3/SBFspot.cfg
Reading config '/usr/local/bin/sbfspot.3/SBFspot.cfg'
Tue Feb 25 14:35:01 2020: INFO: Starting...
sunrise: 07:37
sunset : 18:09
Connecting to 00:80:25:25:43:50 (1/10)
Initializing...
SUSyID: 125 - SessionID: 849595210 (0x32A3CB4A)
SMA netID=02
Serial Nr: xxxxxxxxxxxxxxxxx
BT Signal=71.0%
Logon OK
Local Host Time: 25/02/2020 14:35:01
Plant Time     : 25/02/2020 14:35:02 (+1 sec)
TZ offset      : 3600 sec - DST: On
Last Time Set  : 25/02/2020 07:30:03
Time was already adjusted today
SUSyID: 131 - xxxxxxxxxxx
Device Name:      xxxxxxxxxx
Device Class:     PV-omvormers
Device Type:      SB 3000HF-30
Software Version: 02.54.11.R
Serial number:    xxxxxxxxxxx
SUSyID: 131 - xxxxxxxxxxxx
Device Status:      OK
SUSyID: 131 - xxxxxxxxxxxxxxxx
Device Temperature: 41.5°C
SUSyID: 131 - xxxxxxxxxxxxxx
GridRelay Status:      Gesloten
SUSyID: 131 - xxxxxxxxxxxxx
Pac max phase 1: 3000W
Pac max phase 2: 3000W
Pac max phase 3: 3000W
SUSyID: 131 - xxxxxxxxxxxxx
Energy Production:
        EToday: 5.120kWh
        ETotal: 19264.981kWh
        Operation Time: 17823.69h
        Feed-In Time  : 16698.38h
SUSyID: 131 - xxxxxxxxxxxxxxxxx
DC Spot Data:
        String 1 Pdc:   2.352kW - Udc: 280.18V - Idc:  8.398A
        String 2 Pdc:   0.000kW - Udc:   0.00V - Idc:  0.000A
SUSyID: 131 - xxxxxxxxxxxxxx
AC Spot Data:
        Phase 1 Pac :   2.236kW - Uac: 236.10V - Iac:  9.474A
        Phase 2 Pac :   0.000kW - Uac:   0.00V - Iac:  0.000A
        Phase 3 Pac :   0.000kW - Uac:   0.00V - Iac:  0.000A
        Total Pac   :   2.236kW
SUSyID: 131 - xxxxxxxxxxxxxxxx
Grid Freq. : 49.98Hz
SUSyID: 131 - xxxxxxxxxxxx
Current Inverter Time: 25/02/2020 14:34:59
Inverter Wake-Up Time: 25/02/2020 07:35:08
Inverter Sleep Time  : 25/02/2020 14:35:02
ExportSpotDataToCSV()
********************
* ArchiveDayData() *
********************
startTime = 5E545570 -> 25/02/2020 00:00:00
ExportDayDataToCSV()
Tue Feb 25 14:35:05 2020: INFO: Done.
SBFspot V3.5.2

table vwPvoData shows lines like:
TimeStamp	Name	Type	Serial	V1	V2	V3	V4	V5	V6	V7	V8	V9	V10	V11	V12	PVoutput
2020-02-25 17:55:00	SN: 2120170067	SB 3000HF-30	2120170067	19266433	24			28,8	226,25							
2020-02-25 17:50:00	SN: 2120170067	SB 3000HF-30	2120170067	19266431	36			29	225,15							
2020-02-25 17:45:00	SN: 2120170067	SB 3000HF-30	2120170067	19266428	48			29	228,13							
2020-02-25 17:40:00	SN: 2120170067	SB 3000HF-30	2120170067	19266424	72			29,1	227,78

berbergh, 2020-02-26:
Sorry, I made a mistake to close the subject unintentionaly.

saylinglow, 2020-02-26:
berbergh - remember UNIX/Linux is case sensitive and grep sbfspot will give different results to grep SBFspot

sbf-, 2020-02-26:
What is the output of
`SELECT * FROM [vwPvoData] where pvoutput is null order by timestamp desc;`

berbergh, 2020-02-26:
saylinglow - 

> 
> 
> berbergh - remember UNIX/Linux is case sensitive and grep sbfspot will give different results to grep SBFspot

That is why I gave you both.

berbergh, 2020-02-26:
> SELECT * FROM [vwPvoData] where pvoutput is null order by timestamp desc;

Weird. It says Error: database or disk is full

Does this mean the database is corrupt?

sbf-, 2020-02-26:
Database or SD card corrupt. Similar issue like #303

berbergh, 2020-02-26:
I run on Raspian.

sbf-, 2020-02-26:
> I run on Raspian.

So?

berbergh, 2020-02-26:
Is there a way to safely put back a copy?

berbergh, 2020-02-26:
> 
> 
> > I run on Raspian.
> 
> So?

I guess I am a humble user, who does not know how to use windows applications in a Raspian environment.
Anyhow, I made a copy of the SBFspot.db to windows and started SQLite there. I Also found a way to run the PRAGMA integrity_check, but all I got was OK.

Then I stopped the cron job and the daemon and created a new database.
Now, I see new data appear in PVoutput.

Now I wonder, is there a way to read old data into the new database?

sbf-, 2020-02-26:
SBFspot -? For all options

closed 2020-04-06 by sbf-