SBFspot/SBFspot #369: Inquires often not completed / running forever

99technetium, 2020-02-26:
Hi all,

for some reason inquiries often do not complete. Instead SBFspot seems to wait for more packets but only receives data from the SMA Sunny Home Manager 2.0 until I terminate the run. As a workaround for cron, I'm using the timeout command to terminate / kill an inquiry if they does not exit within a time frame. By the log files and the "Start"/"Done" messages if have determined a average ratio of 55 % (37-68%) of successful inquiries over a time of 2.5 days and ~ 1300 inquiries. If an inquiry is complete, the data is also successfully inserted into the SQL db.

I have attached two log files (-d5 -v5), one for a successful and one for a incomplete inquiry. The ip 192.168.6.110 is the Sunny Home Manager, 192.168.6.111 the inverter. 

Can this behavior be caused by some sort of packet loss due to UDP and Sunny Home Manager sending every second on the same Multicast? This happens with both, the solar and battery inverter.

I tried inquiring both inverters with one call and also separately with two calls and two config files. Both variants show the same issue but separate configs seem to be more successful. If i inquiry both inverters within one call, the first one often completes but the second one fails with the described behavior and runs forever

Script to call SBFspot (one config for each inverter):

`#!/bin/bash`
`log=/var/log/sbfspot.3/Battery_$(date '+%Y%m%d').log`
`timeout -k 60 30 SBFspot -v -ad1 -am0 -ae0 -finq -cfg<config>.cfg >> $log`

[read_stuck.log](https://github.com/SBFspot/SBFspot/files/4256496/read_stuck.log)
[read_successful.log](https://github.com/SBFspot/SBFspot/files/4256497/read_successful.log)

sbf-, 2020-02-26:
Well, I think the solution here is to ignore the HM data.

99technetium, 2020-02-26:
Just tried blocking the HM by iptables. Seems to do the job. Is there an option in SBFspot to ignore specific IPs? Would like to use the HM as well.

sbf-, 2020-03-05:
> Is there an option in SBFspot to ignore specific IPs?

No (not yet)

sbf-, 2020-03-09:
Maybe this can help: #371

99technetium, 2020-03-09:
I altered the two lines in the latest release (3.6.0) and so far it seems to be working fine. I’ll check again in the next few days, but I'm confident because this problem occurred every few inquiries before.

In any case, thank you very much for your great work and thanks to @oelmaier for this fix!

sellth, 2020-04-09:
This also solved my similar problem with a Sunny Tripower 6.0 inverter.

99technetium, 2020-04-09:
Sorry for the late feedback, it's running perfectly fine after applying #371. Thanks again! 
@sbf- are you going to merge oelmalers commits into master?

sbf-, 2020-04-09:
Thanks for testing this!
I'll merge #371 asap

sbf-, 2020-06-20:
This is fixed in V3.7.0

closed 2020-06-20 by sbf-