SBFspot/SBFspot #370: SBFUpdate not sending data

auror1968, 2020-03-04:
Hi

Recently installed 3.6.0 on Ubuntu 18.04.4 LTS; data is added to the SQLlite database fine. The uploader  doesn't put anything into PVoutput. In the logfile it just says the Daemon is starting and nothing else.

I can't see any errors anywhere, I changed the logging in upload cfg to Debug but get nothing else.

Thanks in advance for any help.

Nigel

sbf-, 2020-03-05:
Some things to check:

date/time of inverter correct?

`sudo systemctl status SBFspotUpload`

`SELECT * FROM [vwPvoData] where pvoutput is null order by timestamp desc;`

auror1968, 2020-03-06:
Thanks for coming back.

- Checked the clock ad it is correct.

- sudo systemctl status SBFspotUpload

This is the output:

 SBFspotUpload.service - SBFspot Upload Daemon
   Loaded: loaded (/usr/local/bin/sbfspot.3/SBFspotUpload.service; enabled; vendor preset: enabled)
   Active: active (running) since Tue 2020-03-03 16:47:13 UTC; 2 days ago
 Main PID: 10754 (SBFspotUploadDa)
    Tasks: 1 (limit: 1109)
   CGroup: /system.slice/SBFspotUpload.service
           └─10754 /usr/local/bin/sbfspot.3/SBFspotUploadDaemon

Mar 03 16:47:13 pvoutput systemd[1]: Started SBFspot Upload Daemon.

- SELECT * FROM [vwPvoData] where pvoutput is null order by timestamp desc
Out put is:

2020-02-29 09:10:00|The Garwoods Inverter|SB 2500|2001761517|25451079|156|||||||||||
2020-02-29 09:05:00|The Garwoods Inverter|SB 2500|2001761517|25451066|132|||||||||||
2020-02-29 09:00:00|The Garwoods Inverter|SB 2500|2001761517|25451055|144|||||||||||
2020-02-29 08:55:00|The Garwoods Inverter|SB 2500|2001761517|25451043|132|||||||||||
2020-02-29 08:50:00|The Garwoods Inverter|SB 2500|2001761517|25451032|108|||||||||||
2020-02-29 08:45:00|The Garwoods Inverter|SB 2500|2001761517|25451023|132|||||||||||
2020-02-29 08:40:00|The Garwoods Inverter|SB 2500|2001761517|25451012|144|||||||||||
2020-02-29 08:35:00|The Garwoods Inverter|SB 2500|2001761517|25451000|120|||||||||||
2020-02-29 08:30:00|The Garwoods Inverter|SB 2500|2001761517|25450990|72|||||||||||
2020-02-29 08:25:00|The Garwoods Inverter|SB 2500|2001761517|25450984|48|||||||||||
2020-02-29 08:20:00|The Garwoods Inverter|SB 2500|2001761517|25450980|48|||||||||||
2020-02-29 08:15:00|The Garwoods Inverter|SB 2500|2001761517|25450976|48|||||||||||
2020-02-29 08:10:00|The Garwoods Inverter|SB 2500|2001761517|25450972|72|||||||||||
2020-02-29 08:05:00|The Garwoods Inverter|SB 2500|2001761517|25450966|96|||||||||||
2020-02-29 08:00:00|The Garwoods Inverter|SB 2500|2001761517|25450958|84|||||||||||
2020-02-29 07:55:00|The Garwoods Inverter|SB 2500|2001761517|25450951|84|||||||||||
2020-02-29 07:50:00|The Garwoods Inverter|SB 2500|2001761517|25450944|36|||||||||||
2020-02-29 07:45:00|The Garwoods Inverter|SB 2500|2001761517|25450941|0|||||||||||
2020-02-29 07:40:00|The Garwoods Inverter|SB 2500|2001761517|25450941|12|||||||||||
2020-02-29 07:35:00|The Garwoods Inverter|SB 2500|2001761517|25450940|0|||||||||||
2020-02-29 07:30:00|The Garwoods Inverter|SB 2500|2001761517|25450940|0|||||||||||
2020-02-29 07:25:00|The Garwoods Inverter|SB 2500|2001761517|25450940|0|||||||||||
2020-02-29 07:20:00|The Garwoods Inverter|SB 2500|2001761517|25450940|0|||||||||||
2020-02-29 07:15:00|The Garwoods Inverter|SB 2500|2001761517|25450940|0|||||||||||

sbf-, 2020-03-08:
Strange, data looks OK, except the missing spotdata (Uac,...)

Here I have another query (like it is executed by upload daemon)
``` SQL
SELECT strftime('%Y%m%d,%H:%M',TimeStamp) DTTM,V1,V2,V3,V4,V5,V6,V7,V8,V9,V10,V11,V12
FROM [vwPvoData]
WHERE TimeStamp>DATE(DATE(),'-14 day')
    AND PVoutput IS NULL
    AND Serial=2001761517
ORDER BY TimeStamp
LIMIT 30;
```

auror1968, 2020-03-08:
This is the output I get:

20200229,07:15|25450940|0||||||||||
20200229,07:20|25450940|0||||||||||
20200229,07:25|25450940|0||||||||||
20200229,07:30|25450940|0||||||||||
20200229,07:35|25450940|0||||||||||
20200229,07:40|25450941|12||||||||||
20200229,07:45|25450941|0||||||||||
20200229,07:50|25450944|36||||||||||
20200229,07:55|25450951|84||||||||||
20200229,08:00|25450958|84||||||||||
20200229,08:05|25450966|96||||||||||
20200229,08:10|25450972|72||||||||||
20200229,08:15|25450976|48||||||||||
20200229,08:20|25450980|48||||||||||
20200229,08:25|25450984|48||||||||||
20200229,08:30|25450990|72||||||||||
20200229,08:35|25451000|120||||||||||
20200229,08:40|25451012|144||||||||||
20200229,08:45|25451023|132||||||||||
20200229,08:50|25451032|108||||||||||
20200229,08:55|25451043|132||||||||||
20200229,09:00|25451055|144||||||||||
20200229,09:05|25451066|132||||||||||
20200229,09:10|25451079|156||||||||||
20200229,09:15|25451094|180||||||||||
20200229,09:20|25451113|228||||||||||
20200229,09:25|25451133|240||||||||||
20200229,09:30|25451156|276||||||||||
20200229,09:35|25451181|300||||||||||
20200229,09:40|25451207|312||||||||||

sbf-, 2020-03-09:
Forgot the obvious... What is in the logfile?
Depending of the setup in `/home/pi/smadata/logs` or `/var/log/sbfspot.3`

auror1968, 2020-03-10:
pi@pvoutput:/var/log/sbfspot.3$ cat SBFspotUpload20200303.log
[16:47:12] INFO: Stopping Daemon...
[16:47:13] INFO: SBFspotUploadDaemon Version 2.0.0
[16:47:13] INFO: Starting Daemon...

I only have a SBFspotUploadxxx.log file for the date I started the Daemon, should there be one for each day ?

Have attached log files also
[MyPlant_20200310.log](https://github.com/SBFspot/SBFspot/files/4311415/MyPlant_20200310.log)
[SBFspotUpload20200303.log](https://github.com/SBFspot/SBFspot/files/4311416/SBFspotUpload20200303.log)

sbf-, 2020-03-10:
> I only have a SBFspotUploadxxx.log file for the date I started the Daemon, should there be one for each day ?

No, it only starts a new log when it writes something (which it doesn't)
Something must be wrong in SBFspotUpload.cfg. Can you post the content?
Don't forget to remove the API key ;-)

auror1968, 2020-03-10:
################################################################################
#  SBFspotUpload.cfg - Configuration file for SBFspotUploadService/Daemon
#  (c)2012-2018, SBF (https://github.com/SBFspot/SBFspot)
#
#  DISCLAIMER:
#  A user of SBFspotUploadService/Daemon software acknowledges that he or she is
#  receiving this software on an "as is" basis and the user is not relying on
#  the accuracy or functionality of the software for any purpose. The user
#  further acknowledges that any use of this software will be at his own risk
#  and the copyright owner accepts no responsibility whatsoever arising from
#  the use or application of the software.
#
################################################################################

################################
### Log Settings             ###
################################
# Windows: C:\Users\Public\SMAdata\Logs
# Linux  : /var/log/sbfspot.3
#LogDir=C:\Users\Public\SMAdata\Logs
LogDir=/var/log/sbfspot.3

#LogLevel=debug|info|warning|error (default info)
LogLevel=debug

################################
### PVoutput Upload Settings ###
################################
#PVoutput_SID (PVoutput_System_ID)
#Map inverters to PVoutput System ID's
#PVoutput_SID=SerialNmbrInverter_1:PVoutput_System_ID_1,SerialNmbrInverter_2:PVoutput_System_ID_2
#e.g. PVoutput_SID=200212345:4321
PVoutput_SID=00000000:18320

#PVoutput_Key
#Sets PVoutput API Key
PVoutput_Key=xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx

################################
### SQL DB Settings          ###
################################
# SQL_Database (Fullpath to SQLite DB)
# Windows: C:\Users\Public\SMAdata\SBFspot.db
# Linux  : /home/pi/smadata/SBFspot.db
#SQL_Database=C:\Users\Public\SMAdata\SBFspot.db
SQL_Database=/home/pi/smadata/SBFspot.db

# Reserved for MySQL
#SQL_Database=SBFspot
#SQL_Hostname=<Network Name> or <IP-address>
#SQL_Username=SBFspotUser
#SQL_Password=SBFspotPassword

sbf-, 2020-03-10:
Serial nr is missing!!!
`PVoutput_SID=00000000:18320`
should be
`PVoutput_SID=2001761517:18320`

However, I doubt about `SID=18320` as it points to another system ([Caithness Road](https://pvoutput.org/intraday.jsp?sid=18320))
And last but not least, since you installed it on Ubuntu, I think the path to the db will be wrong too:
`SQL_Database=/home/pi/smadata/SBFspot.db`

auror1968, 2020-03-11:
That was it put the serial number in and it now works. 
Sorry you can't cater for idiots :-)

I created a user called PI so I didn't need to change paths etc.

Many thanks for this great tool

closed 2020-03-11 by auror1968