SBFspot/SBFspot #372: SBFspot Grid Power stuck to 0

Albert1G, 2020-03-24:
Dera all, 
I'm new here, so I apologize if this issue was noticed before, I searched but here I'm, to asking you some help.
I'm running SBFspot V3.5.2 on Raspberry with Raspbian GNU/Linux 10 (buster).
I have a SMA inverter  SB3.0-1AV-40 085, and everything works almost well.

I have a problem reading the AC spot data:, Phase1 ac is 0, but Total Pac is right (compared with the real data reads by SMA web interface)

 ````
[...]
DC Spot Data:
        String 1 Pdc:   0.231kW - Udc: 134.50V - Idc:  1.720A
        String 2 Pdc:   0.358kW - Udc: 183.70V - Idc:  1.950A
SUSyID: 342 - SN: 1992108085
AC Spot Data:
        Phase 1 Pac :   0.000kW - Uac: 237.17V - Iac:  2.420A
        Phase 2 Pac :   0.000kW - Uac:   0.00V - Iac:  0.000A
        Phase 3 Pac :   0.000kW - Uac:   0.00V - Iac:  0.000A
        Total Pac   :   0.559kW
SUSyID: 342 - SN: 1992108085
Grid Freq. : 49.99Hz
SUSyID: 342 - SN: 1992108085
Current Inverter Time: 24/03/2020 08:56:19
Inverter Sleep Time  : 24/03/2020 08:56:19
ExportSpotDataToCSV()
[...]
```
Where am I wrong?
I need the Phase 1 Pac value to pass it to 123solar (SBFspot -123s=DATA)

Thanks
Ciao
Albert

saylinglow, 2020-03-24:
As far as I can see, the special code for the 123solar logger (in CSVexport.cpp) only calculates total Pac, total Iac and maximum Uac. The Pdc1, Pdc2, Pdc3 and Pac1, Pac2, Pac3 values sent are the ones as they were collected from the inverter. I'm not sure if this is a shortcoming in the code, or an enhancement.

Albert1G, 2020-03-24:
Negletting the 123solar and -123s option 
The command     SBFspot -v     gives:
-----------
DC Spot Data:
        String 1 Pdc:   0.839kW - Udc: 122.30V - Idc:  6.860A
        String 2 Pdc:   1.512kW - Udc: 224.40V - Idc:  6.730A
SUSyID: 342 - SN: 1992108085
AC Spot Data:
        Phase 1 Pac :   0.000kW - Uac: 248.36V - Iac:  9.960A
        Phase 2 Pac :   0.000kW - Uac:   0.00V - Iac:  0.000A
        Phase 3 Pac :   0.000kW - Uac:   0.00V - Iac:  0.000A
        Total Pac   :   2.289kW
-------------
- Total Pac ->  ok
- Phase 1 Pac -> = 0

It seems a SBFspot "overall" problem not releted to 123s option.

saylinglow, 2020-03-24:
Albert1G, in the SBFspot.cfg file, what do you have the parameter 'CalculateMissingSpotValues' set to? It should be set to 1 so that SBFspot calculates the power values if they are not given by the inverter, I think.

Albert1G, 2020-03-24:
Dear Saylinglow, 
was set to 0; now I set it on 1. Now everything is going great!
Many many thanks!
Ciao

closed 2020-03-24 by Albert1G