SBFspot/SBFspot #373: SBFspot stopped working after upgrading to Raspbian Buster

jvlerick, 2020-03-25:
After upgrading my system to buster, SBFspot no longer reports anything to my Domoticz.
When testing by running "/usr/local/bin/sbfspot.3/SBFspot -v -finq -nocsv" I get this error: "error while loading shared libraries: libboost_date_time.so.1.62.0: cannot open shared object file: No such file or directory". 
Apparently with upgrading to Buster libboost version is now 1.67. 
I went through all the installation steps( [https://github.com/SBFspot/SBFspot/wiki/Installation-Linux-SQLite](url) again but no change.
Is there any way to solve this or is SBFspot not compatible with Buster?

fonske, 2020-03-25:
You need to do the recompile SBFspot en SBFspotupload daemon steps again after upgrading to Buster
I had the same issues yesterday.

jvlerick, 2020-03-25:
I am not very experienced with Linux. Can you explain how to recompile SBFspot? I do not upload data to PVoutput so no need for the SBFspotUploadDaemon

sbf-, 2020-03-25:
You have a few options:
* Recompile SBFspot from scratch by following the [wiki](https://github.com/SBFspot/SBFspot/wiki/Installation-Linux-SQLite)
* Extracting SBFspot from the [precompiled files](https://github.com/SBFspot/SBFspot/releases/tag/V3.5.2) and replacing the existing SBFspot in /usr/local/bin/sbfspot.3
This executable is statically linked with BOOST 1.66 and doesn't care about installed BOOST version

jvlerick, 2020-03-26:
Thanks. In the meantime I had indeed already re-installed from scratch. Just going through the steps of the wiki did not work (as mentioned in my first post). So I removed all relevant directories first and started again.

deagel, 2020-03-26:
hi,
look like #255

closed 2020-03-26 by sbf-