SBFspot/SBFspot #376: Calculation of DayData wattage is very large

RossWilliamson, 2020-04-04:
Recently the DayData wattage value is extremely high. I put a few debug statements into the code and it looks like it's reading a large value initially with only the final read giving an equivalent of the total watt/hours (see below).  I'm wondering if the memory is full on the sunnyboy or something?

DC Spot Data:
	String 1 Pdc:   0.236kW - Udc: 213.25V - Idc:  1.107A
	String 2 Pdc:   0.318kW - Udc: 273.38V - Idc:  1.165A
SUSyID: 247 - SN: 1913120348
AC Spot Data:
	Phase 1 Pac :   0.275kW - Uac: 123.04V - Iac:  2.241A
	Phase 2 Pac :   0.276kW - Uac: 123.19V - Iac:  2.241A
	Phase 3 Pac :   0.000kW - Uac:   0.00V - Iac:  0.000A
	Total Pac   :   0.551kW
SUSyID: 247 - SN: 1913120348
Grid Freq. : 59.96Hz
SUSyID: 247 - SN: 1913120348
Current Inverter Time: 03/04/2020 17:47:24
Inverter Wake-Up Time: 03/04/2020 17:34:23
Inverter Sleep Time  : 03/04/2020 17:47:26
ExportSpotDataToCSV()
********************
* ArchiveDayData() *
********************
startTime = 5E86DEF0 -> 03/04/2020 00:00:00
TEST:packetcount: 5
TEST:(x):41:totalWH:18446744073709551615
TEST:(x):53:totalWH:18446744073709551615
.... Lot's of repeated values through to packetcount: 0
TEST:(x):497:totalWH:18446744073709551615
TEST:packetcount: 0
TEST:(x):41:totalWH:18446744073709551615
TEST:(x):53:totalWH:18446744073709551615
TEST:(x):65:totalWH:18446744073709551615
TEST:(x):77:totalWH:18446744073709551615
TEST:(x):89:totalWH:18446744073709551615
TEST:(x):101:totalWH:18446744073709551615
TEST:(x):113:totalWH:18446744073709551615
TEST:(x):125:totalWH:18446744073709551615
TEST:(x):137:totalWH:18446744073709551615
TEST:(x):149:totalWH:18446744073709551615
TEST:(x):161:totalWH:18446744073709551615
TEST:(x):173:totalWH:18446744073709551615
TEST:(x):185:totalWH:18446744073709551615
TEST:(x):197:totalWH:18446744073709551615
TEST:(x):209:totalWH:18446744073709551615
TEST:(x):221:totalWH:18446744073709551615
TEST:(x):233:totalWH:18446744073709551615
TEST:(x):245:totalWH:18446744073709551615
TEST:(x):257:totalWH:18446744073709551615
TEST:(x):269:totalWH:32459356
ExportDayDataToCSV()
**********************
* ArchiveMonthData() *
**********************
startTime = 5E84F2C0 -> 01/04/2020 13:00:00
ExportMonthDataToCSV()
There is no data to export!
Reading events: 2020-Apr-01
ExportEventsToCSV()
Fri Apr  3 17:47:29 2020: INFO: Done.

notreallybob, 2020-04-04:
Looks like arithmetic overflow.

sbf-, 2020-04-04:
18446744073709551615 or 0xFFFFFFFFFFFFFFFF is the representation for NaN (Not a Number)
It looks like there is no data available. Can you provide a debuglog with -d5?

RossWilliamson, 2020-04-05:
See attached debug log.  Note I'm on a RPI-3.  This issue started around mid-march.  Note that at the same time the sunny portal info report email also showed no data.... But it's clear that I can connect, the instantaneous power looks good as do the voltages etc... I'm wondering if a "power-cycle" is needed on the invetor

[debug.log](https://github.com/SBFspot/SBFspot/files/4432909/debug.log)

sbf-, 2020-04-05:
>I'm wondering if a "power-cycle" is needed on the invertor
> 
If it started all of a sudden I think it’s a good idea to start with a reset

closed 2020-04-27 by sbf-

RossWilliamson, 2020-05-07:
Just to add closure to this it was a hardware fault.  The display unit was replaced and all is well now