SBFspot/SBFspot #380: MQTT login with username and password

MarnikC, 2020-04-13:
When creating an MQTT connection it is not possible to set a Username and Password for the MQTT server.

sbf-, 2020-04-13:
Add username and password to the MQTT_PublisherArgs option in SBFspot.cfg

MarnikC, 2020-04-13:
This solves the login problem on the MQTT server. Thanks

sbf-, 2021-07-15:
See #473

sbf-, 2021-12-28:
Don't remember why I reopened this ;-)

closed 2021-12-28 by sbf-

nubbler, 2023-09-28:
I tried also now to setup mqtt with password, but it didn't work....

`MQTT: Publishing (sbfspot) Timestamp": "28/09/2023 09:20:51""SunRise": "28/09/2023 07:22:00""SunSet": "28/09/2023 19:11:00""InvSerial": 2100075453"InvName": "SN: 2100075453""InvStatus": "Ok""EToday": 2.006"ETotal": 89852.302"PACTot": 2692.000"UDC1": 355.280"UDC2": 359.730"IDC1": 3.858"IDC2": 3.965"PDC1": 1370.000"PDC2": 1426.000
Connection Refused: not authorised.
Error: The connection was refused.
MQTT: Failed te execute '/usr/bin/mosquitto_pub' mosquitto client installed?
Error 1280 while publishing to MQTT Broker`

I I run

mosquitto_sub -d -h <IP> -t sbfspot -u <user> -P <password>

it runs and the system connects to mqtt broker.

but with SBFSpot it didn't work.

SBFspot.cfg:

`#########################
###   MQTT Settings   ###
#########################

# Full path to mosquitto_pub executable
# Linux: sudo apt-get install mosquitto-clients
MQTT_Publisher=/usr/bin/mosquitto_pub

# Windows: Download and install mosquitto clients from https://mosquitto.org/download/
# Mosquitto 64-bit on Windows 64-bit
# MQTT_Publisher=%ProgramW6432%\\mosquitto\\mosquitto_pub.exe

# Mosquitto 32-bit on Windows 32-bit
# MQTT_Publisher=%ProgramFiles%\\mosquitto\\mosquitto_pub.exe

# IP address or DNS name of MQTT Broker
# Don't use test broker for production environments
MQTT_Host=<ip>
# MQTT_Host=test.mosquitto.org
# MQTT_Host=broker.hivemq.com

# Port (Optional)
#MQTT_Port=

# Topic Name
# When using public broker like test.mosquitto.org make sure to define a unique topic name
# by appending the serial number of the inverter to avoid you receive data of another SBFspot user. 
MQTT_Topic=sbfspot_openhabian
#MQTT_Topic=sbfspot_{plantname}/sma_{serial}

# Format of message items to be sent
# JSON: MQTT_ItemFormat="{key}": {value}
# TEXT: MQTT_ItemFormat={key}:{value}
# XML:  MQTT_ItemFormat=<item name="{key}" value="{value}" />
MQTT_ItemFormat="{key}": {value}

# MQTT_ItemDelimiter (none|blank|comma|semicolon default comma)
# JSON: MQTT_ItemDelimiter=comma
# TEXT: MQTT_ItemDelimiter=semicolon
# XML : MQTT_ItemDelimiter=none
MQTT_ItemDelimiter=comma

# Mandatory arguments for mosquitto_pub executable
# JSON: MQTT_PublisherArgs=-h {host} -t {topic} -m "{{message}}"
# TEXT: MQTT_PublisherArgs=-h {host} -t {topic} -m "{message}"
# XML : MQTT_PublisherArgs=-h {host} -t {topic} -m "<mqtt_message>{message}</mqtt_message>"
MQTT_PublisherArgs=-h {host} -t {topic} -r -m "{{message}}" -u <user> -P <password>`

Because running mosquitto_sub in comand line works I think mqtt is set up and working well.
So something in SBFspot.cfg is wrong? But I don't find a mistake?

Did I anything missed?