SBFspot/SBFspot #381: Continual Error Loops - "Power value [XXXXXXXX] too high for system size [3640]

bundy67, 2020-04-18:
Hello, 
I have been through the forums to note people have had similar problems that I am seeing when  uploading to PVOutput and getting an error message like this

**14:06:07] ERROR: Uploading 3 datapoints, starting with 20200418,12:55,36961287,443535456,,,36.2,244.9 Bad request 400: Power value [443535456] too high for system size [3640]**

So I followed the Instructions on these forums, connected to sqlite and deleted the erroneous value using the following command

First I connect to the database
**sqlite3 /home/pi/smadata/SBFspot.db**

Then run the following command
sqlite> **select * from daydata where power >= 7000;**
Response is 
**1587185700|2130085479|36961287|443535456|**
I then run 
**delete from daydata where power >= 7000;**

I run the select command again to confirm the value has been deleted and it has. NB: I chose 7000 as the value as my system size is 3640. I have tried running using the exact value as in the database "443535456" but I get the same issue.

So far so good, the bad entry is deleted (so it seems) and next time the system tries to update PVoutput the following message is in the log and I can see on PVoutput that data has updated. So at this point it is working again

**[14:15:09] INFO: Uploading datapoint: 20200418,14:10,36961639,480,,,36,246.04 => OK (200)**

However this is when things get weird.....

The next time it tries to run a scheduled update I see the exact same error message I started out with

**[14:17:04] ERROR: Uploading 2 datapoints, starting with 20200418,12:55,36961287,443535456,,,36.2,244.9 Bad request 400: Power value [443535456] too high for system size [3640]**

If I again delete using sqlite it starts to update once more (just a single update works) then immediately after that update the error is back and I am continually facing the same issue and going around in circles. 

Any idea what I can do to purge this once and for all? My system has been running flawlessly for 18 months and stopped updating around 4 weeks ago. I'm only now had time to look into it and at the moment am stuck on what is my next move

Thanks in advance

sbf-, 2020-04-18:
Change the pvodata view as described in #137

bundy67, 2020-04-18:
Thanks for the reply. Just to be clear do you mean use DB Browser to go to the table pvodata (or vwPvoData) and edit the values there?

There are a few comments in the thread you refer to so I am not sure of the exact steps I should be taking and in what table(s) I should be looking at

thanks again

sbf-, 2020-04-18:
No, you can't change the data in vwPvoData. With this workaround you leave the wrong data in the db, but you just ignore it for the upload.

PVoutput allows a max power value=150% of the system size. In your case (3640W) this is appox. 5460W

You can change the vwPvodata view to limit the powervalue to 5400 (to be on the safe side)
```SQL
CASE WHEN dd.Power > 5400
    THEN 0
    ELSE dd.Power
END
```

The comlete view looks like this: (SQLite has no "ALTER VIEW" so you have to DROP it first.)
```SQL
DROP VIEW IF EXISTS vwPvoData;

CREATE VIEW vwPvoData AS
       SELECT dd.Timestamp,
              dd.Name,
              dd.Type,
              dd.Serial,
              dd.TotalYield AS V1,
              CASE WHEN dd.Power > 5400
              THEN 0
              ELSE dd.Power
              END AS V2,
              cons.EnergyUsed AS V3,
              cons.PowerUsed AS V4,
              spot.Temperature AS V5,
              spot.Uac1 AS V6,
              NULL AS V7,
              NULL AS V8,
              NULL AS V9,
              NULL AS V10,
              NULL AS V11,
              NULL AS V12,
              dd.PVoutput
         FROM vwDayData AS dd
              LEFT JOIN vwAvgSpotData AS spot
                     ON dd.Serial = spot.Serial AND dd.Timestamp = spot.Nearest5min
              LEFT JOIN vwAvgConsumption AS cons
                     ON dd.Timestamp = cons.Nearest5min
        ORDER BY dd.Timestamp DESC;
```

sbf-, 2020-04-18:
> The next time it tries to run a scheduled update I see the exact same error message I started out with
> 
> **[14:17:04] ERROR: Uploading 2 datapoints, starting with 20200418,12:55,36961287,443535456,,,36.2,244.9 Bad request 400: Power value [443535456] too high for system size [3640]**

This is correct behaviour: SBFspot is storing missing (the record you just deleted) data in the db. If you have SunnyExplorer (Windows only) installed, you can check the data there. I don't think it's an SBFspot bug, but your inverter reporting bad data.

bundy67, 2020-04-19:
Ok thank you, I have checked via Sunny Explorer and can not see any huge numbers (at least in the Graphs ) but not to worry. 

This may be pushing it and excuse my lack of knowledge but I presume I need to run both of the Scripts as shown above?

I've tried to Execute the SQL from within DB Browser but I am not really sure what I am doing to be honest and it doesn't seem to fix things. If there's some reading to guide me would appreciate any pointers. 

Alternatively (and this may be an easier option for me) 

Would reinstalling SBFSpot software from scratch be worthwhile and give me a clean database getting rid of that bad record?

bundy67, 2020-04-19:
Just an update my system now seems to be working. I'm not sure if it was me or just luck. What I did was add both those SQL scripts into DB Browser in the "Execute SQL" tab. 

I hit the Execute Button and also wrote that to the database and rebooted my system and so far (1 hour in) it has been updating PVoutput without errors. 

Just for the next person who comes along was this the right thing to do? Or did I just get lucky? :-) 

Thanks again for all your help and providing such a great tool

sbf-, 2020-05-29:
I'll try to tackle this recurring issue in one of the next releases

commit 527b2a4b (2021-12-11): Fix various issues Bad request 400: Power value too high for system size:
Fix #137
Fix #313
Fix #381
Fix #384
closed 2020-05-29 by sbf-