SBFspot/SBFspot #382: Error: [battery_data]mysql_stmt_execute() returned : Field 'Extra' doesn't have a default value

rucksman, 2020-04-20:
Since some days I get this error message:  

> Error: [battery_data]mysql_stmt_execute() returned : Field 'Extra' doesn't have a default value

On some days there is no such message at all, on other days, there are 2 or 3, or sometimes more than 10. Also the time when this message occurs do vary: in the morning, arround midday, in the afternoon, on the evening, during the night, ... So no clear pattern for me in order to have an idea what might happen.

Can you help?

SBFspot version 3.6, Inverter SB-3600SE.

sbf-, 2020-04-20:
Strange, since the current table definition doesn't have an "Extra" Field
``` SQL
CREATE TABLE SpotDataX (
    `TimeStamp` INTEGER (4) NOT NULL,
    `Serial`    INTEGER (4) NOT NULL,
    `Key`       INTEGER (4) NOT NULL,
    `Value`     INTEGER (4),
    PRIMARY KEY (
        `TimeStamp` ASC,
        `Serial` ASC,
        `Key`
    )
);
```
It was there before in some test phase long ago...
Did you upgrade to 3.6 recently?

The best thing to do is to alter the table and drop the "extra" field. There should be no data in it.

rucksman, 2020-04-21:
My current version was compiled January 20th (v3.6), the code was downloaded January 9th, The only modification I made was the one related to issue #350.

sbf-, 2020-07-08:
'Extra' field should be dropped

closed 2020-07-08 by sbf-