SBFspot/SBFspot #384: No data received at pvoutput

ivansanders, 2020-04-24:
Hi there, have been uploading for over a week without issue.   This morning though, there is no data uploaded.

Raspbian (Buster)

Have done three checks:
Database, Upload daemon, and cronlog.   

Appreciate any guidance

Thanks
Ivan

**pi@raspberrypi:~ $ /usr/local/bin/sbfspot.3/SBFspot -finq -v5 -d**
SBFspot V3.5.2
Yet another tool to read power production of SMA solar inverters
(c) 2012-2018, SBF (https://github.com/SBFspot/SBFspot)
Compiled for Linux (LE) 32 bit with SQLite support
Commandline Args: -finq -v5 -d
Reading config '/usr/local/bin/sbfspot.3/SBFspot.cfg'
Configuration settings:
IP_Address=0.0.0.0
Password=<undisclosed>
MIS_Enabled=0
Plantname=45delprat
OutputPath=/home/pi/smadata/%Y
OutputPathEvents=/home/pi/smadata/%Y/Events
Latitude=-37.81
Longitude=144.96
Timezone=Australia/Melbourne
CalculateMissingSpotValues=0
DateTimeFormat=%d/%m/%Y %H:%M:%S
DateFormat=%d/%m/%Y
TimeFormat=%H:%M:%S
SynchTime=1
SynchTimeLow=1
SynchTimeHigh=3600
SunRSOffset=900
DecimalPoint=comma
CSV_Delimiter=semicolon
Precision=3
CSV_Export=1
CSV_ExtendedHeader=1
CSV_Header=1
CSV_SaveZeroPower=1
CSV_Spot_TimeSource=0
CSV_Spot_WebboxHeader=0
Locale=en-US
BTConnectRetries=10
SQL_Database=/home/pi/smadata/SBFspot.db
### End of Config ###
Fri Apr 24 10:24:23 2020: INFO: Starting...
sunrise: 06:57
sunset : 17:39
Connecting to Local Network...
Initializing...
SUSyID: 125 - SessionID: 822650471 (0x3108A667)
--------: 00 01 02 03 04 05 06 07 08 09
00000000: 53 4D 41 00 00 04 02 A0 FF FF 
00000010: FF FF 00 00 00 20 00 00 00 00 
20 Bytes sent to IP [239.12.255.254]
MAX_CommBuf is now 60 bytes
Received 60 bytes from IP [192.168.1.114]
--------: 00 01 02 03 04 05 06 07 08 09
00000000: 53 4D 41 00 00 04 02 A0 00 00 
00000010: 00 01 00 02 00 00 00 01 00 04 
00000020: 00 10 00 01 00 03 00 04 00 20 
00000030: 00 00 00 01 00 04 00 30 C0 A8 
00000040: 01 72 00 04 00 40 00 00 00 00 
00000050: 00 02 00 70 EF 0C 00 00 00 00 
Inverter IP address: 192.168.1.114 found via broadcastidentification
[... 420 lines cut ...]
54 Bytes sent to IP [192.168.1.114]
Fri Apr 24 10:24:25 2020: INFO: Done.
pi@raspberrypi:~ $ 

**pi@raspberrypi:~ $ sudo systemctl status SBFspotUpload**
● SBFspotUpload.service - SBFspot Upload Daemon
   Loaded: loaded (/usr/local/bin/sbfspot.3/SBFspotUpload.service; enabled; vendor preset: enable
   Active: active (running) since Fri 2020-04-24 08:29:06 AEST; 1h 56min ago
 Main PID: 433 (SBFspotUploadDa)
    Tasks: 1 (limit: 4035)
   Memory: 7.3M
   CGroup: /system.slice/SBFspotUpload.service
           └─433 /usr/local/bin/sbfspot.3/SBFspotUploadDaemon

Apr 24 08:29:06 raspberrypi systemd[1]: Started SBFspot Upload Daemon.
lines 1-10/10 (END)

**cronlog :**
Apr 24 09:40:01 raspberrypi CRON[6284]: (pi) CMD (/usr/local/bin/sbfspot.3/daydata)
Apr 24 09:45:01 raspberrypi CRON[6658]: (pi) CMD (/usr/local/bin/sbfspot.3/daydata)
Apr 24 09:50:01 raspberrypi CRON[7032]: (pi) CMD (/usr/local/bin/sbfspot.3/daydata)
Apr 24 09:55:01 raspberrypi CRON[7421]: (pi) CMD (/usr/local/bin/sbfspot.3/daydata)
Apr 24 10:00:01 raspberrypi CRON[7827]: (pi) CMD (/usr/local/bin/sbfspot.3/daydata)
Apr 24 10:04:24 raspberrypi crontab[8146]: (pi) LIST (pi)
Apr 24 10:05:01 raspberrypi CRON[8203]: (pi) CMD (/usr/local/bin/sbfspot.3/daydata)

sbf-, 2020-04-24:
I’m working on a [wiki](https://github.com/SBFspot/SBFspot/wiki/No-uploads-to-PVOutput.org) to address this issue. Not finished yet, but you can have a look already.

ivansanders, 2020-04-24:
Cheers, thats a huge help

[08:55:31] ERROR: Uploading 30 datapoints, starting with 20200424,01:10,1839193,0 Bad request 400: Power value [22070328] too high for system size [6600]

Any idea what that might be about?

sbf-, 2020-04-24:
Sure, see #137 and #381

sbf-, 2020-04-24:
Take a look at the power value of 06:30 in your log above. This happens sometimes in the startup phase of the inverter.

ivansanders, 2020-04-24:
Right!! That will do it!  Sorry for being a noob - is that just a matter of deleting that row out of SBFspot.db?   I don't suppose that will be going into your wiki any time soon? ;-)

sbf-, 2020-04-25:
No, you should change the view vwPVOdata as described in the workaround.

ivansanders, 2020-04-25:
Well that was a bit of a learning curve! - but thanks, all fixed :)

commit 527b2a4b (2021-12-11): Fix various issues Bad request 400: Power value too high for system size:
Fix #137
Fix #313
Fix #381
Fix #384
closed 2020-04-26 by sbf-