SBFspot/SBFspot #386: Creation of Spot files inconsistent

coderph0x, 2020-05-01:
Hello!

First of all, thanks for the great work on this project!

I have two issues that I need some explanation on. Maybe you can clarify or lead me to a source that does:

1. Although spot data is uploaded to the MySQL DB without issues, the spot data in den csv file only gets populated maybe once an hour. Why is this?

2. I have some old spot files from before I used the MySQL DB. Is there some sort of uploader, that does conveniently convert from the csv files to sql?

Best regards
Marco

sbf-, 2020-05-01:
When enabled, CSV files are created/updated every time SBFspot runs. Set verbose logging to see if the CSV is locked for writing.
Importing of old files is not yet possible.

coderph0x, 2020-05-01:
Will do. Since the files get updated irregularly, it's somewhat strange but maybe the logs sure something.
Does the MySQL uploader do anything different than pushing the csv values into the tables? If not, i will probably be able to write a script that does the trick.

coderph0x, 2020-05-01:
Well, the logs and the verbose output don't show 
any irregularities, but the spot data files get updated approx. once an hour, sometimes twice. I can't figure it out, are there some constraint checks in the code, that might prohibit the writing of the csv file?

sbf-, 2020-07-08:
Is this issue still valid?

If you have more than 1 device (e.g. solar inverter and battery inverter) there might be a problem when the order of connection to the devices is changed.

coderph0x, 2020-07-08:
I just checked the logs and the bugs still exists. Spot data gets uploaded to SQL every 2 minutes but the csv files only gets updated every 10 to 30, sometimes 60, minutes. It's only one inverter and a pretty unmodified install.

sbf-, 2021-04-18:
Unable to reproduce

closed 2021-04-18 by sbf-