SBFspot/SBFspot #390: sbfspot and mysql/mariadb

tpn59, 2020-06-02:
Hi everybody,
I have an issue with sbfspot and mysql:
mariadb 5 runs on a synology diskstation, sbfspot is installed on a raspi4. 
I ran sbfspot-config and everything goes fine up to contacting the database. 
Ran the two .sql scripts and the database and -user were created. 
A connect to the db as the database user goes well. 
But: If I run SBFspot -v  as  test, I get an access denied error for SBFspotUser. 
All DB privileges are OK. 
Any ideas what to do?
Thanks dir your efforts, regards
Thomas

sbf-, 2020-06-03:
Review the SBFspotUser privileges. Currently CreateMySQLUser.sql is for MySQL/MariaDB running on the same RPi. This is not the case in your situation:
``` SQL
CREATE USER 'SBFspotUser'@'localhost' IDENTIFIED BY 'SBFspotPassword';
GRANT DELETE,INSERT,SELECT,UPDATE ON SBFspot.* TO 'SBFspotUser'@'localhost';
FLUSH PRIVILEGES;
```
Replace localhost with the DNS name or IP of your RPi (or use % to have access from any device)
```
CREATE USER 'SBFspotUser'@'%' IDENTIFIED BY 'SBFspotPassword';
GRANT DELETE,INSERT,SELECT,UPDATE ON SBFspot.* TO 'SBFspotUser'@'%';
FLUSH PRIVILEGES;
```

tpn59, 2020-06-03:
Hi,
thanks al lot. I missed out to look at that .sql file.
Now it runs.
Regards
Thomas

closed 2020-06-19 by sbf-