SBFspot/SBFspot #392: Data/graphics to high solar panels

Blobke, 2020-06-15:
Hi

I am new to SBFspot and when checking my grapics for my solar panels i see a hugh spike and i have no idea where it comes from.

[image]

checking the .CSV file i can't see anything wrong and in the DB in vwdaydata everything looks fine at first sight.

Anyone an idea where this is coming from?

Blobke, 2020-06-15:
Hi

Here is a more detail screenshot

[image]

sbf-, 2020-06-15:
From where do you take the data for the graphs? CSV or SQL DB?
If you see it in the graph, you should see it in the datasource too.

Blobke, 2020-06-16:
Hi

I have checked the .CSV files and the DB but i can't explain the spike

[SBFspot.zip](https://github.com/SBFspot/SBFspot/files/4785974/SBFspot.zip)

I have added the CSV & the DB? Maybe someone can see what is wrong?

sbf-, 2020-06-17:
There is nothing wrong in the db/csv so the problem is in your graphical tool.
What is the data source of your graph?

Blobke, 2020-06-17:
Hi

where can i see what data source the graph is taking?

sbf-, 2020-06-17:
You don't know and we don't know ;-)
We don't even know what visualization software you're using.
This is not SBFspot related, so I'll close this issue

closed 2020-06-17 by sbf-

Blobke, 2020-06-19:
Hi

Yes i have found where the issue is and it is in the domoticz.db, the meter table.
It is recording there with 2 digits to much.

I have posted this in the domoticz forum.

https://www.domoticz.com/forum/viewtopic.php?f=6&t=33107

thx anyway for the help.