SBFspot/SBFspot #394: Compile error on V3.7.0, Centos 8 & mysql

DJE007, 2020-06-21:
make mysql
test -d mysql/obj/ || mkdir -p mysql/obj/
test -d mysql/bin/ || mkdir -p mysql/bin/
g++ boost_ext.cpp -c -Wall -O2 -Wno-unused-local-typedefs -DUSE_MYSQL -o mysql/obj/boost_ext.o
g++ misc.cpp -c -Wall -O2 -Wno-unused-local-typedefs -DUSE_MYSQL -o mysql/obj/misc.o
g++ sunrise_sunset.cpp -c -Wall -O2 -Wno-unused-local-typedefs -DUSE_MYSQL -o mysql/obj/sunrise_sunset.o
g++ SBFNet.cpp -c -Wall -O2 -Wno-unused-local-typedefs -DUSE_MYSQL -o mysql/obj/SBFNet.o
g++ CSVexport.cpp -c -Wall -O2 -Wno-unused-local-typedefs -DUSE_MYSQL -o mysql/obj/CSVexport.o
g++ Ethernet.cpp -c -Wall -O2 -Wno-unused-local-typedefs -DUSE_MYSQL -o mysql/obj/Ethernet.o
g++ EventData.cpp -c -Wall -O2 -Wno-unused-local-typedefs -DUSE_MYSQL -o mysql/obj/EventData.o
g++ ArchData.cpp -c -Wall -O2 -Wno-unused-local-typedefs -DUSE_MYSQL -o mysql/obj/ArchData.o
g++ SBFspot.cpp -c -Wall -O2 -Wno-unused-local-typedefs -DUSE_MYSQL -o mysql/obj/SBFspot.o
g++ TagDefs.cpp -c -Wall -O2 -Wno-unused-local-typedefs -DUSE_MYSQL -o mysql/obj/TagDefs.o
g++ Bluetooth.cpp -c -Wall -O2 -Wno-unused-local-typedefs -DUSE_MYSQL -o mysql/obj/Bluetooth.o
g++ mqtt.cpp -c -Wall -O2 -Wno-unused-local-typedefs -DUSE_MYSQL -o mysql/obj/mqtt.o
g++ db_MySQL.cpp -c -Wall -O2 -Wno-unused-local-typedefs -DUSE_MYSQL -o mysql/obj/db_MySQL.o
g++ db_MySQL_Export.cpp -c -Wall -O2 -Wno-unused-local-typedefs -DUSE_MYSQL -o mysql/obj/db_MySQL_Export.o
g++ mysql/obj/boost_ext.o mysql/obj/misc.o mysql/obj/sunrise_sunset.o mysql/obj/SBFNet.o mysql/obj/CSVexport.o mysql/obj/Ethernet.o mysql/obj/EventData.o mysql/obj/ArchData.o mysql/obj/SBFspot.o mysql/obj/TagDefs.o mysql/obj/Bluetooth.o mysql/obj/mqtt.o mysql/obj/db_MySQL.o mysql/obj/db_MySQL_Export.o -s -o mysql/bin/SBFspot -Wl,-Bdynamic -lpthread -lbluetooth -lboost_date_time -lmysqlclient
mysql/obj/SBFspot.o: In function `GetConfig(Config*)':
SBFspot.cpp:(.text+0x75c4): undefined reference to `boost::system::system_category()'
SBFspot.cpp:(.text+0x766b): undefined reference to `boost::system::system_category()'
SBFspot.cpp:(.text+0x77b8): undefined reference to `boost::system::system_category()'
SBFspot.cpp:(.text+0x77d4): undefined reference to `boost::system::system_category()'
SBFspot.cpp:(.text+0x77f8): undefined reference to `boost::system::system_category()'
mysql/obj/SBFspot.o:SBFspot.cpp:(.text+0x79df): more undefined references to `boost::system::system_category()' follow
mysql/obj/SBFspot.o: In function `boost::system::error_category::std_category::equivalent(std::error_code const&, int) const':
SBFspot.cpp:(.text._ZNK5boost6system14error_category12std_category10equivalentERKSt10error_codei[_ZNK5boost6system14error_category12std_category10equivalentERKSt10error_codei]+0x30): undefined reference to `boost::system::generic_category()'
SBFspot.cpp:(.text._ZNK5boost6system14error_category12std_category10equivalentERKSt10error_codei[_ZNK5boost6system14error_category12std_category10equivalentERKSt10error_codei]+0xa9): undefined reference to `boost::system::generic_category()'
SBFspot.cpp:(.text._ZNK5boost6system14error_category12std_category10equivalentERKSt10error_codei[_ZNK5boost6system14error_category12std_category10equivalentERKSt10error_codei]+0x121): undefined reference to `boost::system::generic_category()'
mysql/obj/SBFspot.o: In function `boost::system::error_category::std_category::equivalent(int, std::error_condition const&) const':
SBFspot.cpp:(.text._ZNK5boost6system14error_category12std_category10equivalentEiRKSt15error_condition[_ZNK5boost6system14error_category12std_category10equivalentEiRKSt15error_condition]+0x27): undefined reference to `boost::system::generic_category()'
SBFspot.cpp:(.text._ZNK5boost6system14error_category12std_category10equivalentEiRKSt15error_condition[_ZNK5boost6system14error_category12std_category10equivalentEiRKSt15error_condition]+0x61): undefined reference to `boost::system::generic_category()'
mysql/obj/SBFspot.o: In function `_GLOBAL__sub_I_MAX_CommBuf':
SBFspot.cpp:(.text.startup+0x2d33): undefined reference to `boost::system::system_category()'
collect2: error: ld returned 1 exit status
make: *** [makefile:119: mysql/bin/SBFspot] Error 1

DJE007, 2020-06-21:
Changed makefile to;

$(TARGET): $(OBJECTS)
	$(LD) $^ $(LDFLAGS) -o $@ -Wl,-lboost_system,-Bdynamic $(addprefix -l,$(LIBS)) $(addprefix -L,$(LIBDIR))

make clean && make mysql
rm -rf nosql
rm -rf sqlite
rm -rf mysql
rm -rf mariadb
test -d mysql/obj/ || mkdir -p mysql/obj/
test -d mysql/bin/ || mkdir -p mysql/bin/
g++ boost_ext.cpp -c -Wall -O2 -Wno-unused-local-typedefs -DUSE_MYSQL -o mysql/obj/boost_ext.o
g++ misc.cpp -c -Wall -O2 -Wno-unused-local-typedefs -DUSE_MYSQL -o mysql/obj/misc.o
g++ sunrise_sunset.cpp -c -Wall -O2 -Wno-unused-local-typedefs -DUSE_MYSQL -o mysql/obj/sunrise_sunset.o
g++ SBFNet.cpp -c -Wall -O2 -Wno-unused-local-typedefs -DUSE_MYSQL -o mysql/obj/SBFNet.o
g++ CSVexport.cpp -c -Wall -O2 -Wno-unused-local-typedefs -DUSE_MYSQL -o mysql/obj/CSVexport.o
g++ Ethernet.cpp -c -Wall -O2 -Wno-unused-local-typedefs -DUSE_MYSQL -o mysql/obj/Ethernet.o
g++ EventData.cpp -c -Wall -O2 -Wno-unused-local-typedefs -DUSE_MYSQL -o mysql/obj/EventData.o
g++ ArchData.cpp -c -Wall -O2 -Wno-unused-local-typedefs -DUSE_MYSQL -o mysql/obj/ArchData.o
g++ SBFspot.cpp -c -Wall -O2 -Wno-unused-local-typedefs -DUSE_MYSQL -o mysql/obj/SBFspot.o
g++ TagDefs.cpp -c -Wall -O2 -Wno-unused-local-typedefs -DUSE_MYSQL -o mysql/obj/TagDefs.o
g++ Bluetooth.cpp -c -Wall -O2 -Wno-unused-local-typedefs -DUSE_MYSQL -o mysql/obj/Bluetooth.o
g++ mqtt.cpp -c -Wall -O2 -Wno-unused-local-typedefs -DUSE_MYSQL -o mysql/obj/mqtt.o
g++ db_MySQL.cpp -c -Wall -O2 -Wno-unused-local-typedefs -DUSE_MYSQL -o mysql/obj/db_MySQL.o
g++ db_MySQL_Export.cpp -c -Wall -O2 -Wno-unused-local-typedefs -DUSE_MYSQL -o mysql/obj/db_MySQL_Export.o
g++ mysql/obj/boost_ext.o mysql/obj/misc.o mysql/obj/sunrise_sunset.o mysql/obj/SBFNet.o mysql/obj/CSVexport.o mysql/obj/Ethernet.o mysql/obj/EventData.o mysql/obj/ArchData.o mysql/obj/SBFspot.o mysql/obj/TagDefs.o mysql/obj/Bluetooth.o mysql/obj/mqtt.o mysql/obj/db_MySQL.o mysql/obj/db_MySQL_Export.o -s -o mysql/bin/SBFspot -Wl,-lboost_system,-Bdynamic -lpthread -lbluetooth -lboost_date_time -lmysqlclient

DJE007, 2020-06-21:
Working fine with changes

SBFspot V3.7.0
Yet another tool to read power production of SMA solar inverters
(c) 2012-2020, SBF (https://github.com/SBFspot/SBFspot)
Compiled for Linux (LE) 64 bit with MySQL support
Commandline Args: -v5
Reading config '/usr/local/SMAspot/SBFspot.cfg'
Configuration settings:
BTAddress=00:80:25:1D:EE:40
MIS_Enabled=0
Plantname=Homer
OutputPath=/usr/local/SMAspot/OUT/%Y
OutputPathEvents=/usr/local/SMAspot/Events/%Y
Latitude=-27.8638
Longitude=153.288
Timezone=Australia/Brisbane
CalculateMissingSpotValues=1
DateTimeFormat=%d/%m/%Y %H:%M:%S
DateFormat=%d/%m/%Y
TimeFormat=%H:%M:%S
SynchTime=7
SynchTimeLow=1
SynchTimeHigh=3600
SunRSOffset=900
DecimalPoint=comma
CSV_Delimiter=semicolon
Precision=3
CSV_Export=0
CSV_ExtendedHeader=1
CSV_Header=1
CSV_SaveZeroPower=1
CSV_Spot_TimeSource=0
CSV_Spot_WebboxHeader=0
Locale=en-US
BTConnectRetries=10
SQL_Database=SBFspot
SQL_Hostname=localhost
SQL_Username=SBFspotUser
### End of Config ###
Mon Jun 22 07:04:21 2020: INFO: Starting...
sunrise: 06:39
sunset : 16:58
Connecting to 00:80:25:1D:EE:40 (1/10)
Initializing...
SUSyID: 125 - SessionID: 968915979 (0x39C07C0B)
SMA netID=01
Serial Nr: 7D32CF3A (2100481850)
BT Signal=69.4%
Logon OK
Local Host Time: 22/06/2020 07:04:22
Plant Time     : 22/06/2020 07:04:21 (-1 sec)
TZ offset      : 36000 sec - DST: Off
Last Time Set  : 22/06/2020 07:03:34
Time was already adjusted in last 7 days
SUSyID: 78 - SN: 2100481850
Device Name:      Homer
Device Class:     Solar Inverters
Device Type:      SB 4000TL-20
Software Version: 03.20.05.R
Serial number:    2100481850
SUSyID: 78 - SN: 2100481850
Device Status:      Ok
SUSyID: 78 - SN: 2100481850
Device Temperature: 47.6°C
SUSyID: 78 - SN: 2100481850
GridRelay Status:      Closed
SUSyID: 78 - SN: 2100481850
Pac max phase 1: 4000W
Pac max phase 2: 0W
[... 117 lines cut ...]
* ArchiveMonthData() *
**********************
startTime = 5ED46120 -> 01/06/2020 12:00:00
SUSyID: 78 - SN: 2100481850
01/06/2020 : 40478.330kWh - 15.395kWh
02/06/2020 : 40491.829kWh - 13.499kWh
03/06/2020 : 40502.371kWh - 10.542kWh
04/06/2020 : 40514.950kWh - 12.579kWh
05/06/2020 : 40528.271kWh - 13.321kWh
06/06/2020 : 40535.257kWh - 6.986kWh
07/06/2020 : 40543.852kWh - 8.595kWh
08/06/2020 : 40555.703kWh - 11.851kWh
09/06/2020 : 40563.930kWh - 8.227kWh
10/06/2020 : 40574.179kWh - 10.249kWh
11/06/2020 : 40587.669kWh - 13.490kWh
12/06/2020 : 40597.664kWh - 9.995kWh
13/06/2020 : 40600.128kWh - 2.464kWh
14/06/2020 : 40612.357kWh - 12.229kWh
15/06/2020 : 40623.323kWh - 10.966kWh
16/06/2020 : 40634.481kWh - 11.158kWh
17/06/2020 : 40641.033kWh - 6.552kWh
18/06/2020 : 40651.163kWh - 10.130kWh
19/06/2020 : 40657.897kWh - 6.734kWh
20/06/2020 : 40669.775kWh - 11.878kWh
======

tdk32, 2020-06-25:
Thanks for your solution! Works fine under Debian Buster x64.

`Changed makefile to;

$(TARGET): $(OBJECTS)
$(LD) $^ $(LDFLAGS) -o $@ -Wl,**-lboost**_system,-Bdynamic $(addprefix -l,$(LIBS)) $(addprefix -L,$(LIBDIR))

make clean && make mariadb`

sbf-, 2020-06-26:
I think it's better to keep libs together:
```
LIBS       := pthread bluetooth boost_date_time boost_system
```

sbf-, 2020-06-29:
@geert-hendrickx-be 
I had a closer look at the issue and found [this PR](https://github.com/SBFspot/SBFspot/pull/361/commits/9f833837414d2d5c59799dfbb43de2288a486f1b) is the culprit
The make file is missing the boost_system lib

geert-hendrickx-be, 2020-06-29:
I cannot reproduce that failure, its builds fine for me, both natively on Arch Linux x64_64 with mysql as well as cross building for OpenWRT mips with sqlite.

I don't see boost::system being referenced anywhere?

(OpenWRT packages all boost runtime libraries separately to save embedded diskspace, which is why I tried to minimize those dependencies.)

sbf-, 2020-06-29:
See [StackOverflow](https://stackoverflow.com/questions/9723793/undefined-reference-to-boostsystemsystem-category-when-compiling)

> Linking with a library that defines the missing symbol (-lboost_system) is the obvious solution, but in the particular case of Boost.System, a misfeature in the original design makes it use boost::system::generic_category() and boost::system::system_category() needlessly. Compiling with the flag -DBOOST_SYSTEM_NO_DEPRECATED disables that code and lets a number of programs compile without requiring -lboost_system (that link is of course still needed if you explicitly use some of the library's features).
Starting from Boost 1.66 and this commit, this behavior is now the default, so hopefully fewer and fewer users should need this answer.
As noticed by @AndrewMarshall, an alternative is to define BOOST_ERROR_CODE_HEADER_ONLY which enables a header-only version of the code. This was discouraged by Boost as it can break some functionality. However, since 1.69, header-only seems to have become the default, supposedly making this question obsolete.

No need to link with `boost_system`, but compile with `-DBOOST_ERROR_CODE_HEADER_ONLY`

```
CFLAGS := -c -Wall -O2 -Wno-unused-local-typedefs -DBOOST_ERROR_CODE_HEADER_ONLY
```

geert-hendrickx-be, 2020-06-29:
Ok, that explains; I was building with boost 1.71, but centos 8 has 1.66.

DJE007, 2020-07-14:
Thanks SBF your hard work is appreciated

sbf-, 2021-01-05:
Fixed in [71fcaeb](https://github.com/SBFspot/SBFspot/commit/71fcaeb9e9de0382d90941313e40d006d7959517)

closed 2021-01-05 by sbf-