SBFspot/SBFspot #395: Mix of ISO-8859-1 and UTF-8, CRLF and LF line endings

geert-hendrickx-be, 2020-06-29:
Hi

Currently the repository has a mix of files with ISO-8859-1 (latin-1) and UTF-8 encodings, as well as CRLF ("DOS") and LF ("UNIX") line endings, even both newline styles within files.

Could you standardize on one file encoding?  Preferably UTF-8 and LF. ;-)

sbf-, 2020-06-30:
Development is done with Visual Studio on Windows, so CR/LF is default. Don't know where single LF comes from, but VS has no problems with it ;-)
Same for encodings, no idea how this happened, but again VS knows how to handle... maybe some other tools used, like WinMerge.
Never had issues with these different encodings...

geert-hendrickx-be, 2020-06-30:
Config files (and .sql templates etc) installed on Linux should be LF terminated, but that's platform specific.  Maybe let the makefile run `dos2unix` over the textfiles in the install target?

For the encoding you may want to keep the code and config files plain ASCII, as it's mostly just the ® sign here and there...

For the TagLists the encoding matters for the console output, but it should take the system locale encoding in consideration for that.

sbf-, 2021-01-13:
Hmm...

```
"c:\Program Files\git\usr\bin\file.exe" *
ArchData.cpp:            C source, ISO-8859 text, with CRLF line terminators
ArchData.h:              C source, Non-ISO extended-ASCII text, with CRLF, LF line terminators
Bluetooth.cpp:           C source, UTF-8 Unicode text, with CRLF, LF line terminators
CMakeLists.txt:          ASCII text, with CRLF line terminators
CSVexport.cpp:           C source, ISO-8859 text, with very long lines, with CRLF, LF line terminators
CSVexport.h:             C source, Non-ISO extended-ASCII text, with CRLF, LF line terminators
CreateMySQLDB.sql:       ASCII text, with CRLF line terminators
CreateMySQLUser.sql:     ASCII text, with CRLF line terminators
CreateSQLiteDB.sql:      ASCII text, with CRLF line terminators
Ethernet.cpp:            C source, Non-ISO extended-ASCII text, with CRLF, LF line terminators
Ethernet.h:              C source, Non-ISO extended-ASCII text, with CRLF line terminators
EventData.cpp:           C source, Non-ISO extended-ASCII text, with CRLF, LF line terminators
EventData.h:             C++ source, Non-ISO extended-ASCII text, with CRLF, LF line terminators
Rec40Att.h:              C++ source, Non-ISO extended-ASCII text, with CRLF, LF line terminators
Rec40S32.h:              C++ source, Non-ISO extended-ASCII text, with CRLF, LF line terminators
SBFNet.cpp:              C source, UTF-8 Unicode text, with CRLF line terminators
SBFNet.h:                C source, Non-ISO extended-ASCII text, with CRLF line terminators
SBFspot.cfg:             ISO-8859 text, with CRLF line terminators
SBFspot.cpp:             C source, ISO-8859 text, with CRLF line terminators
SBFspot.h:               C source, ISO-8859 text, with CRLF line terminators
SBFspot.vcxproj:         XML 1.0 document, UTF-8 Unicode (with BOM) text, with CRLF line terminators
SBFspot.vcxproj.filters: XML 1.0 document, UTF-8 Unicode (with BOM) text, with CRLF line terminators
SQLselect.h:             C source, Non-ISO extended-ASCII text, with CRLF, LF line terminators
TagDefs.cpp:             C source, Non-ISO extended-ASCII text, with CRLF, LF line terminators
TagDefs.h:               C++ source, Non-ISO extended-ASCII text, with CRLF, LF line terminators
TagListDE-DE.txt:        Non-ISO extended-ASCII text, with CRLF line terminators
TagListEN-US.txt:        Non-ISO extended-ASCII text, with CRLF line terminators
TagListES-ES.txt:        Non-ISO extended-ASCII text, with CRLF line terminators
TagListFR-FR.txt:        Non-ISO extended-ASCII text, with CRLF line terminators
TagListIT-IT.txt:        Non-ISO extended-ASCII text, with CRLF line terminators
TagListNL-NL.txt:        Non-ISO extended-ASCII text, with CRLF line terminators
Types.h:                 C source, UTF-8 Unicode text, with CRLF line terminators
Update_340_MySQL.sql:    ASCII text, with CRLF line terminators
Update_340_SQLite.sql:   ASCII text, with CRLF line terminators
bluetooth.h:             C source, Non-ISO extended-ASCII text, with CRLF line terminators
boost_ext.cpp:           C source, Non-ISO extended-ASCII text, with CRLF, LF line terminators
boost_ext.h:             C source, Non-ISO extended-ASCII text, with CRLF line terminators
date_time_zonespec.csv:  ASCII text, with CRLF line terminators
db_MySQL.cpp:            C source, ISO-8859 text, with CRLF line terminators
db_MySQL.h:              C++ source, Non-ISO extended-ASCII text, with CRLF, LF line terminators
db_MySQL_Export.cpp:     C source, ISO-8859 text, with CRLF, LF line terminators
db_MySQL_Export.h:       C++ source, Non-ISO extended-ASCII text, with CRLF, LF line terminators
db_SQLite.cpp:           C source, ISO-8859 text, with CRLF, LF line terminators
db_SQLite.h:             C++ source, Non-ISO extended-ASCII text, with CRLF, LF line terminators
db_SQLite_Export.cpp:    C source, ISO-8859 text, with CRLF line terminators
db_SQLite_Export.h:      C++ source, Non-ISO extended-ASCII text, with CRLF, LF line terminators
endianness.h:            Non-ISO extended-ASCII text, with CRLF, LF line terminators
main.cpp:                C source, UTF-8 Unicode text, with CRLF line terminators
makefile:                makefile script, ASCII text, with CRLF line terminators
misc.cpp:                C source, Non-ISO extended-ASCII text, with CRLF, LF line terminators
misc.h:                  C source, Non-ISO extended-ASCII text, with CRLF, LF line terminators
mqtt.cpp:                C source, ISO-8859 text, with CRLF line terminators
mqtt.h:                  C source, Non-ISO extended-ASCII text, with CRLF line terminators
oslinux.h:               C source, UTF-8 Unicode text, with CRLF line terminators
osselect.h:              C source, UTF-8 Unicode text, with CRLF line terminators
oswindows.h:             C source, UTF-8 Unicode text, with CRLF, LF line terminators
strptime.cpp:            C source, UTF-8 Unicode text, with CRLF line terminators
sunrise_sunset.cpp:      C source, Non-ISO extended-ASCII text, with CRLF, LF line terminators
sunrise_sunset.h:        C source, Non-ISO extended-ASCII text, with CRLF line terminators
version.h:               ASCII text, with CRLF line terminators
```

commit a5eca269 (2021-01-17): Fix #395 Mix of ISO-8859-1 and UTF-8, CRLF and LF line endings
closed 2021-01-17 by sbf-