SBFspot/SBFspot #396: Simple daily output report

MrMEScott, 2020-07-01:
If anyone is interested, I've written a small Python program that generates a graph of the output and total power numbers for a given day. It runs on the headless Raspberry Pi that I've installed SBFspot on. Example attached.
I made this to provide a simple way to see how my system is performing without any external dependencies (e.g. pvoutput.org etc.)

[image]

saylinglow, 2020-07-01:
Funnily enough, I was thinking of doing something similar! Are you willing to share it?

MrMEScott, 2020-07-01:
Happy to share. It should be pretty self-explanatory to anyone who has a smattering of Python, but otherwise I'd need to write some documentation to explain the few dependencies and how to configure it.

MrMEScott, 2020-07-01:
Attached is a zip containing the program and a readme file.
[sbfchart.zip](https://github.com/SBFspot/SBFspot/files/4859368/sbfchart.zip)

closed 2020-07-05 by MrMEScott