SBFspot/SBFspot #398: SBFspot docker MariaDB10

TerrorSource, 2020-07-08:
Hi,

I'm trying to configure SBFspot in Docker to use MariaDB 10 on a Synology to save my data.
Exporting/importing in MariaDB 10 is much easier as using csv files or a single db file.
Docker image i'm using: https://github.com/nakla/sbfspot

SBFspot.cfg
``# MySQL
SQL_Database=sbfspot
SQL_Hostname=192.168.1.x
SQL_Username=SBFspotUser
SQL_Password=password``

Error via Terminal: 
``Error: Can't open MySQL db [sbfspot] : Can't connect to MySQL server on '192.168.1.x' (115)``

How do i get this working? that SBFspot puts the data in MariaDB 10

sbf-, 2020-07-08:
Maybe related to #7 ?

TerrorSource, 2020-07-08:
Still trying things to make this work.
Trying on a Raspberry Pi now.

1. Switches MariaDB 10 on Synology to port 3306 instead of 3307
2. Change Core settings: https://stackoverflow.com/questions/45254222/getting-error-sqlstatehy000-2002-connection-refused-on-nas-synology
3. Create a user like: 'user'@'clienthostname', so not % or localhost.

I can connect the Pi to the MariaDB 10 on Synology now with "mysql -h 192.168.1.x -u root -p"
I cant however execute the sql files from the script. Had to do the DB.sql by hand
mysql -h 192.168.1.x -p -u sbfspot < /usr/local/bin/sbfspot.3/CreateMySQLDB.sql 

The 'CreateMySQLUser.sql' can't be executed due to the fact that it has a simple password which can't be used on MariaDB 10 on Synology.

TerrorSource, 2020-07-08:
FIXED the part where the user can't be created.
Maybe something to add/change in the script that it is optional instead of forced.

1. wget https://raw.githubusercontent.com/sbfspot/sbfspot-config/master/sbfspot-config
2. mv sbfspot-config sbfspot-config.sh
3. vi sbfspot-config.sh
change:
``cp "$tmp_dir/CreateMySQLUser.sql" "$dir_program"``
to
``#cp "$tmp_dir/CreateMySQLUser.sql" "$dir_program"``
so it doesn't run the sql file.
4. sq!

sbf-, 2020-07-08:
For MariaDB, you are responsible to create the SBFspot database and SBFspotUser because you need to execute the scripts as root.
But maybe this isn't made clear in the script?

An option is to ask for root password to be able to execute the SQL scripts? Not everyone is going to like this, I'm afraid

TerrorSource, 2020-07-08:
I partly agree. 

For MariaDB, the user must execute the scripts manually but why does the script still tries to run it while i choose MariaDB in the gui-part of the script? I guess that’s because the script can run it against MySQL but not against MariaDB while they are under 1 option. 

Maybe split the MySQL/MariaDB in 2 parts and show a message when people choose MariaDB that they need to run the sql files manually up front or create a “pause” part where they can continue when they have run the scripts manually.