SBFspot/SBFspot #400: Compile error on V3.7.0, Raspbian & mysql

KanteTaete, 2020-07-12:
Hello,
I tried to compile SBFspot with mariadb support. Unfortunately, the mysql.h file is not found.
`user@raspberry:~/SBFspot/SBFspot $ sudo make mariadb
test -d mariadb/obj/ || mkdir -p mariadb/obj/
test -d mariadb/bin/ || mkdir -p mariadb/bin/
g++ boost_ext.cpp -c -Wall -O2 -Wno-unused-local-typedefs -DUSE_MYSQL -o mariadb/obj/boost_ext.o
g++ misc.cpp -c -Wall -O2 -Wno-unused-local-typedefs -DUSE_MYSQL -o mariadb/obj/misc.o
g++ sunrise_sunset.cpp -c -Wall -O2 -Wno-unused-local-typedefs -DUSE_MYSQL -o mariadb/obj/sunrise_sunset.o
g++ SBFNet.cpp -c -Wall -O2 -Wno-unused-local-typedefs -DUSE_MYSQL -o mariadb/obj/SBFNet.o
g++ CSVexport.cpp -c -Wall -O2 -Wno-unused-local-typedefs -DUSE_MYSQL -o mariadb/obj/CSVexport.o
g++ Ethernet.cpp -c -Wall -O2 -Wno-unused-local-typedefs -DUSE_MYSQL -o mariadb/obj/Ethernet.o
g++ EventData.cpp -c -Wall -O2 -Wno-unused-local-typedefs -DUSE_MYSQL -o mariadb/obj/EventData.o
g++ ArchData.cpp -c -Wall -O2 -Wno-unused-local-typedefs -DUSE_MYSQL -o mariadb/obj/ArchData.o
g++ SBFspot.cpp -c -Wall -O2 -Wno-unused-local-typedefs -DUSE_MYSQL -o mariadb/obj/SBFspot.o
In file included from SQLselect.h:46,
                 from SBFspot.cpp:72:
db_MySQL.h:41:10: fatal error: mysql/mysql.h: Datei oder Verzeichnis nicht gefunden
 #include <mysql/mysql.h>
          ^~~~~~~~~~~~~~~
compilation terminated.
make: *** [makefile:116: mariadb/obj/SBFspot.o] Fehler 1`

So where should the mysql.h file be located?
I installed mariadb server and client with
`sudo apt-get install mariadb-server mariadb-client `

This issue seems to occur just with V3.7.0 as my prior installation of version 3.6.0 worked.
My OS version is raspbian 10 (buster).

KanteTaete, 2020-07-12:
I found the solution myself as it seems other users had this as well:
[https://www.photovoltaikforum.com/thread/135373-sma-tripower-mit-voksz%C3%A4hler-auslesen/?postID=1991315#post1991315](url)
...
so additionally to `maraidb-server` and `mariadb-client` also the `default-libmysqlclient-dev` needs to be installed with
`sudo apt-get install default-libmysqlclient-dev`
I'm not sure, if also `libmariadb-dev` is needed as it was recommended to be installed in that thread before.
I installed it - but it didn't solve the issue.

So for documentation purposes, I think it's good to describe issue and solution here.
I hope, it helps if others face the same issue.

closed 2020-07-12 by KanteTaete

sbf-, 2020-07-13:
The recommended way to install SBFspot on Raspberry Pi is to use [sbfspot-config](https://github.com/SBFspot/sbfspot-config)

However, if you want to do it the hard way, install mariadb-client and libmariadb-dev
`sudo apt-get install mariadb-client libmariadb-dev`

Create a symbolic link to the directory containing mysql.h
`sudo ln -s /usr/include/mariadb /usr/include/mysql`

KanteTaete, 2021-01-04:
I had to install SBFspot on a new machine, but unfortunately the sbfspot-config script doesn't work as it should.
First, there were issues as the database and user isn't created before running the script, then my existing config file was overwritten (luckily the script creates a backup) and finally an access error to the database was thrown although I checked the user and privileges 3 times.
--> that was when I decided to install SBFspot manually where I run in the same issue again as last year - so I'm happy that I opened that issue ... :-D

But the tool itself (SBFspot) is great!
Thanks for continuously updating / maintaining the tool!