SBFspot/SBFspot #401: multiple strings support

d01cid, 2020-07-15:
when i execute the SBFspot -v3 -finq i have:

**DC Spot Data:
	String 1 Pdc:   0.000kW - Udc:   0.00V - Idc:  0.000A
	String 2 Pdc:   0.000kW - Udc:   0.00V - Idc:  0.000A
AC Spot Data:
	Phase 1 Pac :   0.000kW - Uac:   0.00V - Iac:  0.000A
	Phase 2 Pac :   0.000kW - Uac:   0.00V - Iac:  0.000A
	Phase 3 Pac :   0.000kW - Uac:   0.00V - Iac:  0.000A
	Total Pac   :   0.000kW**

is it hard to add support for more strings?
we need it for the Core1 inverter which have 8 strings but also on some inverters wich have 4 strings.
many thanks,
alain.

sbf-, 2020-07-16:
Wow! 8 strings (Core2 even has 12 strings)
Well, I'm glad I didn't merge #315 yet because changing the SpotData table for that number of strings is NOT the way to go.
I'm working on V3.8 now and I'll try to add it there.

Just to be sure: you see the actual data for string 1/2 (Now all values are 0.00 but I suppose it was already dark)

d01cid, 2020-07-18:
yep, is was dark, but today i did it again @ 14:50 and got following from a Core1:

	**String 1 Pdc:   3.795kW - Udc: 601.12V - Idc:  6.316A
	String 2 Pdc:   3.487kW - Udc: 572.14V - Idc:  6.098A
	Phase 1 Pac :   6.662kW - Uac: 243.05V - Iac: 27.459A
	Phase 2 Pac :   6.724kW - Uac: 244.80V - Iac: 27.532A
	Phase 3 Pac :   6.667kW - Uac: 242.73V - Iac: 27.523A**

as you see, only 2 strings in my output

d01cid, 2020-07-18:
oops, wrong button

sbf-, 2021-04-16:
I think you'll need a stringID too as the order of string data coming in is unknown.
The idea of serialising the data into a blob is not bad at all, but I think it's almost impossible to query the stringdata separately (using SQL) - if that's important?
My idea was to use the SpotdataX table like it is used for battery data, but leaving string 1 & 2 as is in the Spotdata table.
All values in SpotdataX are represented as INT

sbf-, 2021-04-16:
This was my implementation so far:

```
class mppt
{
private:
	int32_t m_Pdc = NaN_S32;
	int32_t m_Udc = NaN_S32;
	int32_t m_Idc = NaN_S32;

public:
	mppt() { }
	~mppt() { }
	
	// Get
	int32_t Pdc() const { return m_Pdc; }
	int32_t Udc() const { return m_Udc; }
	int32_t Idc() const { return m_Idc; }
	float kW() const { return (float)(m_Pdc) / 1000; }
	float Volt() const { return (float)(m_Udc) / 100; }
	float Amp() const { return (float)(m_Idc) / 1000; }

	// Set
	void Pdc(const int32_t Pdc) { m_Pdc = Pdc; }
	void Udc(const int32_t Udc) { m_Udc = Udc; }
	void Idc(const int32_t Idc) { m_Idc = Idc; }

};

struct InverterData
{
    // ...
    std::map<uint8_t, mppt> mpp;
    // ...
};

```

mincequi, 2021-04-16:
Ok, i see. Well, basically the same struct as i have. I will do something with the DB and bring up a solution.
I will let you know.

mincequi, 2021-04-16:
@sbf- , you can check 416ed6fd344d78ab42bb4ebad3379917c85b38e4 for a preliminary version of multi string support.

sbf-, 2021-12-28:
Multi-string inverters are supported in V3.9.0

closed 2021-12-28 by sbf-