SBFspot/SBFspot #402: How to specify bluetooth device on Raspbian?

katastic3, 2020-07-19:
Hello,

I am using SBFspot happily on a Raspberry Pi 3 Model B+.
The Raspberry Pi comes with integrated bluetooth, which is always used, even if an additional usb bluetooth dongle is attached to the raspberry.
The only way I know how to get SBFspot to use the additional usb bluetooth dongle is to disable the integrated bluetooth with adding the line `dtoverlay=pi3-disable-bt` in /boot/config.txt and rebooting the raspberry afterwards.

Unfortunately some of my inverters are at location A, while some others are on location B.
The raspberry is on location A and I'm running an usb cable with integrated repeater chips with the usb bluetooth dongle to location B.

Now I'd like to make SBFspot use the integrated bluetooth on location A inverters and the usb bluetooth dongle on location B inverters.

Unfortunately the only way of achiving this is by replacing the options in /boot/config.txt  with `sed -i '/dtoverlay=pi3-disable-bt/c\#dtoverlay=pi3-disable-bt' /boot/config.txt` or `sed -i '/#dtoverlay=pi3-disable-bt/c\dtoverlay=pi3-disable-bt' /boot/config.txt` and rebooting afterwards with cron.

Suggestions or help appreciated.

sbf-, 2020-07-19:
I have no idea atm

sbf-, 2020-07-19:
I might have a solution...
Find the 2 local BT MAC addresses (I only have 1 available):
```
hcitool dev
Devices:
	hci0	00:1B:DC:00:00:F8
```
Now compile 2 different SBFspot executables with the 2 BT addresses. Change Bluetooth.cpp at line 295:
```
#ifdef linux

int bthConnect(char *btAddr)
{
    struct sockaddr_rc addr = {0};
    struct sockaddr_rc loc_addr = {0};
    int status = 0;
    sock = socket(AF_BLUETOOTH, SOCK_STREAM, BTPROTO_RFCOMM);

    // bind socket to port 1 of the local bluetooth adapter
    loc_addr.rc_family = AF_BLUETOOTH;
    bdaddr_t my_bdaddr = {0xF8, 0x00, 0x00, 0xDC, 0x1B, 0x00};
    loc_addr.rc_bdaddr = my_bdaddr;
    loc_addr.rc_channel = 1;
    status = bind(sock, (struct sockaddr *)&loc_addr, sizeof(loc_addr));

    if (status == 0)
    {
        addr.rc_family = AF_BLUETOOTH;
        addr.rc_channel = 1;
        str2ba(btAddr, &addr.rc_bdaddr);

        // Connect to Inverter
        status = connect(sock, (struct sockaddr *)&addr, sizeof(addr));
    }

    return(status);
}
```
The line `bdaddr_t my_bdaddr = {0xF8, 0x00, 0x00, 0xDC, 0x1B, 0x00};` is important here. It contains the BT addres `00:1B:DC:00:00:F8` in reversed order.

If this works for you, I can include it in the code and make it configurable.

katastic3, 2020-07-19:
Sounds great. I will try and give feedback. Might take some days, because I'm not at home right now.

katastic3, 2020-07-22:
Hello sbf- and thanks for you suggestion.
Tried it today and it works flawlessly so far.
Awesome work!
Configurable integration into sbfspot would be great.

commit 2efcd39d (2021-04-13): Fix #402 How to specify bluetooth device on Raspbian?
closed 2021-04-13 by sbf-