SBFspot/SBFspot #404: suggestion for updating wiki

prein2, 2020-08-04:
since github doesn't allow me to do a pull request on the wiki page:

In the installation section (https://github.com/SBFspot/SBFspot/wiki/Installation-Linux-SQLite)
All the installation steps are written with the pi username. since not everybody is using the pi username is suggest to change that to: instead of /home/pi/.... use ~/...
so for example this:
cd /home/pi/smadata
sqlite3 SBFspot.db < /home/pi/SBFspot/SBFspot/CreateSQLiteDB.sql

becomes this:
cd ~/smadata
sqlite3 SBFspot.db < ~/SBFspot/SBFspot/CreateSQLiteDB.sql

sbf-, 2020-08-06:
What about `sudo chown -R pi:pi /var/log/sbfspot.3` ?
Replace by `sudo chown -R $USER:$USER /var/log/sbfspot.3` Will this always work?

sbf-, 2020-12-14:
Fixed

closed 2020-12-14 by sbf-