SBFspot/SBFspot #405: Installation on windows 10

lb4207, 2020-08-06:
I'm trying to install on windows 10 following the SBFspot.V3.Quick.Reference.Windows.EN.1.1.2.pdf but when I get to the line
"C:\Program Files\SQLite3\sqlite3.exe" SBFspot.db < "C:\Program Files\SBFspot.3\CreateSQLiteDB.sql"
I get the following,
At line:1 char:40
+ "C:\Program Files\SQLite3\sqlite3.exe" SBFspot.db < "C:\Program Files ...
+                                        ~~~~~~~~~~
Unexpected token 'SBFspot.db' in expression or statement.
At line:1 char:51
+ "C:\Program Files\SQLite3\sqlite3.exe" SBFspot.db < "C:\Program Files ...
+                                                   ~
The '<' operator is reserved for future use.
    + CategoryInfo          : ParserError: (:) [], ParentContainsErrorRecordException
    + FullyQualifiedErrorId : UnexpectedToken

and there is no SBFspot_Empty.db in C:\Program Files (x86)\SBFspot.3

sbf-, 2020-08-06:
Running in Powershell instead of cmd.exe?

lb4207, 2020-08-06:
Dooh. Stupid me.
Thank you.

closed 2020-08-06 by sbf-