SBFspot/SBFspot #408: [question] (Comprehensive) Sources on the "SMA Data2+" protocol

geropl, 2020-09-02:
Hi :wave: ,

first things first: Thanks for the effort you put into this! It seems this is the only source of information on the "SMA Data2+" protocol.

And that leads me to my question: How did you get that protocol information? Only by reversing the network traffic? And looking at how the official SMA behave? Or are there any other (more comprehensive) sources on the web?

The official dev sources [here](https://www.sma.de/en/products/sma-developer.html) seem rather limited and geared towards SEMP only. The protocol description of the [meter protocl](https://www.sma.de/fileadmin/content/global/Partner/Documents/SMA_Labs/EMETER-Protokoll-TI-en-10.pdf) looks promising and I can find some similarties but also differences to the one implemented in [the code](https://github.com/SBFspot/SBFspot/blob/master/SBFspot/SBFspot.cpp).

Any advice is greatly appreciated! :pray:

sbf-, 2020-09-02:
Hi,
SBFspot is originally based on Nanode PV monitor.
All the rest is indeed done by examining the traffic between SunnyExplorer and inverter and have a very close look at SunnyExplorer application.
Some clever people helped me a lot with that ;-)
Unfortunately I haven't any documentation I can share, but have a look [here](https://github.com/qurm/python-smadata2/blob/master/doc/protocol.md)
Since the project moved a few times since 2012 (google code -> codeplex -> github) a lot of information (discussion groups) is lost.

geropl, 2020-09-07:
> Unfortunately I haven't any documentation I can share, but have a look here

Thanks for sharing @sbf- . All the links seem to focus on the bluetooth protocol while I'm just interested in Ethernet (as that's my local setup) - at least for now.

To document (for future me, others searching for insight): After some googling (don't recall how I ended up there) I found https://github.com/kettenbach-it/FHEM-SMA-Speedwire/blob/master/76_SMAInverter.pm quite valuable.

closed 2020-09-09 by sbf-