SBFspot/SBFspot #409: Invalid timezone specified:

sulrich70, 2020-09-05:
When I install using "curl -s https://raw.githubusercontent.com/sbfspot/sbfspot-config/master/sbfspot-config | sudo bash"

i connect to two inverters, specify location (which is gets from raspberry pi config) and my lat/long and get this error:

"Reading config '/usr/local/bin/sbfspot.3/SBFspot.cfg'
Invalid timezone specified: Australia/Canberra
Installation aborted"

config reports:
"Running on Raspberry Pi Model B Rev 2
Checking whiptail... Installed.
architecture=armhf
Raspbian GNU/Linux 10 (buster)
username=pi
userhome=/home/pi
Minimum SBFspot version supported by this tool is 3.5.0
Request to install latest version (3.7.0)"

Any ideas?

thank you in advance

sbf-, 2020-09-05:
I added an updated [date_time_zonespec.csv](https://github.com/SBFspot/SBFspot/releases/download/V3.7.0/date_time_zonespec.csv) to [V3.7.0 release](https://github.com/SBFspot/SBFspot/releases/tag/V3.7.0).
Copy it to `/usr/local/bin/sbfspot.3`

sulrich70, 2020-09-05:
Thanks, copied this, Same problem... :-(

SBFspot Configuration Tool V1.1.2
Running on Raspberry Pi Model B Rev 2
Checking whiptail... Installed.
architecture=armhf
Raspbian GNU/Linux 10 (buster)
username=pi
userhome=/home/pi
Minimum SBFspot version supported by this tool is 3.5.0
Request to install latest version (3.7.0)
Searching for existing configuration(s)...
Reading defaults from /var/tmp/sbfspot.3/sbfspot.defaults... Done
Scanning for Bluetooth devices, please wait...
Saving defaults to /var/tmp/sbfspot.3/sbfspot.defaults... Done
Performing some checks before installation... Done
Downloading https://github.com/sbfspot/sbfspot/releases/download/V3.7.0/sbfspot-sqlite-arm-linux-buster.tar.gz
Copying files...
SBFspot.db exists
Checking database...
Schema Version=1
Saving existing SBFspot.cfg... Done
Configuring SBFspot... Done
Running SBFspot...
SBFspot V3.7.0
Yet another tool to read power production of SMA solar inverters
(c) 2012-2020, SBF (https://github.com/SBFspot/SBFspot)
Compiled for Linux (LE) 32 bit with SQLite support
Commandline Args: -v -ad0 -am0 -ae0 -sp0 -nocsv -nosql -finq -cfg/usr/local/bin/sbfspot.3/SBFspot.cfg
Reading config '/usr/local/bin/sbfspot.3/SBFspot.cfg'
Invalid timezone specified: Australia/Canberra
Installation aborted
steve@raspberrypi:~ $ ls /usr/local/bin/sbfspot.3
**date_time_zonespec.csv**  SBFspot.cfg                  SBFspot.cfg.20200905_104907  TagListDE-DE.txt  TagListIT-IT.txt
daydata                 SBFspot.cfg.20200902_161115  SBFspot.cfg.20200905_105142  TagListEN-US.txt  TagListNL-NL.txt
monthdata               SBFspot.cfg.20200902_162152  SBFspot.cfg.20200905_105414  TagListES-ES.txt
SBFspot                 SBFspot.cfg.20200902_162643  SBFspot.cfg.20200906_074928  TagListFR-FR.txt

sbf-, 2020-09-06:
Hmm, I forgot to update the tar files too.
Should be better now ;-)

sulrich70, 2020-09-07:
Nice one. thank you. Says installation complete yay!

commit beef66a0 (2021-01-05): Fix #409: Invalid timezone specified (updated date_time_zonespec.csv)
closed 2021-01-05 by sbf-