SBFspot/SBFspot #410: Compile error on V3.7.0, ubuntu 20.04 & mysql

htttito, 2020-09-18:
Can anyone help?! 

rm -rf nosql
rm -rf sqlite
rm -rf mysql
rm -rf mariadb
test -d mysql/obj/ || mkdir -p mysql/obj/
test -d mysql/bin/ || mkdir -p mysql/bin/
g++ boost_ext.cpp -c -Wall -O2 -Wno-unused-local-typedefs -DUSE_MYSQL -o mysql/obj/boost_ext.o 
g++ misc.cpp -c -Wall -O2 -Wno-unused-local-typedefs -DUSE_MYSQL -o mysql/obj/misc.o 
g++ sunrise_sunset.cpp -c -Wall -O2 -Wno-unused-local-typedefs -DUSE_MYSQL -o mysql/obj/sunrise_sunset.o 
g++ SBFNet.cpp -c -Wall -O2 -Wno-unused-local-typedefs -DUSE_MYSQL -o mysql/obj/SBFNet.o 
g++ CSVexport.cpp -c -Wall -O2 -Wno-unused-local-typedefs -DUSE_MYSQL -o mysql/obj/CSVexport.o 
g++ Ethernet.cpp -c -Wall -O2 -Wno-unused-local-typedefs -DUSE_MYSQL -o mysql/obj/Ethernet.o 
g++ EventData.cpp -c -Wall -O2 -Wno-unused-local-typedefs -DUSE_MYSQL -o mysql/obj/EventData.o 
g++ ArchData.cpp -c -Wall -O2 -Wno-unused-local-typedefs -DUSE_MYSQL -o mysql/obj/ArchData.o 
g++ SBFspot.cpp -c -Wall -O2 -Wno-unused-local-typedefs -DUSE_MYSQL -o mysql/obj/SBFspot.o 
In file included from SQLselect.h:46,
                 from SBFspot.cpp:72:
db_MySQL.h:41:10: fatal error: mysql/mysql.h: Datei oder Verzeichnis nicht gefunden
   41 | #include <mysql/mysql.h>
      |          ^~~~~~~~~~~~~~~
compilation terminated.
make: *** [makefile:116: mysql/obj/SBFspot.o] Fehler 1

sbf-, 2020-09-19:
Did you install `libmysqlclient-dev`?

htttito, 2020-09-19:
No, there is a Problem....
$ sudo apt install libmysqlclient-dev 
Paketlisten werden gelesen... Fertig
Abhängigkeitsbaum wird aufgebaut.       
Statusinformationen werden eingelesen.... Fertig
Einige Pakete konnten nicht installiert werden. Das kann bedeuten, dass
Sie eine unmögliche Situation angefordert haben oder, wenn Sie die
Unstable-Distribution verwenden, dass einige erforderliche Pakete noch
nicht erstellt wurden oder Incoming noch nicht verlassen haben.
Die folgenden Informationen helfen Ihnen vielleicht, die Situation zu lösen:
Die folgenden Pakete haben unerfüllte Abhängigkeiten:
 libmysqlclient-dev : Hängt ab von: libssl-dev soll aber nicht installiert werden
E: Probleme können nicht korrigiert werden, Sie haben zurückgehaltene defekte Pakete.
$ sudo apt install libssl-d
libssl-dev  libssl-doc  
tito@server:~$ sudo apt install libssl-dev 
Paketlisten werden gelesen... Fertig
Abhängigkeitsbaum wird aufgebaut.       
Statusinformationen werden eingelesen.... Fertig
Einige Pakete konnten nicht installiert werden. Das kann bedeuten, dass
Sie eine unmögliche Situation angefordert haben oder, wenn Sie die
Unstable-Distribution verwenden, dass einige erforderliche Pakete noch
nicht erstellt wurden oder Incoming noch nicht verlassen haben.
Die folgenden Informationen helfen Ihnen vielleicht, die Situation zu lösen:
Die folgenden Pakete haben unerfüllte Abhängigkeiten:
 libssl-dev : Hängt ab von: libssl1.1 (= 1.1.1f-1ubuntu2) aber 1.1.1g-1+ubuntu18.04.1+deb.sury.org+1 soll installiert werden
E: Probleme können nicht korrigiert werden, Sie haben zurückgehaltene defekte Pakete.

htttito, 2020-09-20:
I found the solution :-)
- I add the repository of mysql  - https://dev.mysql.com/downloads/repo/apt/
- sudo apt update
- sudo apt install libmysqlclient-dev

And anything works fine :-)

closed 2020-09-20 by htttito