SBFspot/SBFspot #412: skipping unresponsive inverter

cav-oem, 2020-09-25:
Hello,
been using SBFspot for the last 1 year, on a ubuntu server VM.
I have 4 SMA inverters on the network and all has been running great.
Even made a php page that grabs the data from the db for display.

But, a few days back, one of the inverters stopped responding to network requests (either from SBFspot or even ping), probably broken comm board on the inverter.

I had SBFspot configured with fixed ips at the config
> IP_Address=192.168.0.180,192.168.0.181,192.168.0.182,192.168.0.183

has since the start, but the daydata cron query started failing with error: 

> ERROR: Connection to inverter failed!
Is 192.168.0.183 the correct IP?
Please check IP_Address in SBFspot.cfg!

and just exits out not saving any data.
Tried changing to "discovery mode" (0.0.0.0 and even blank) instead of fixed ips but I always get same error but with the inverters found (the good ones).
>ERROR: Connection to inverter failed!
Is 192.168.0.180 the correct IP?
Please check IP_Address in SBFspot.cfg!

I'm guessing discovery mode would work on a clean db instead of an existing one.

**My solution was to just comment out the bad inverter to get the data to flow again.**

However, wasnt it supposed to skip the bad inverter and save the remaining data?

The _Inverters_ table still has the last communication from the inverter, along with the new data from the working inverters.
[image]

Updated to V3.6.0 thinking it could be a previous bug, but still the same.
(V3.7.0 failed to "make sqlite" step due to some "boost::system" errors. RIP me :satisfied:)

The skipping a bad fixed ip inverter might be worth looking into.
Maybe write null values to the db (could also be usefull to trigger alarms by checking via php script).
The serial column is not-null though.

sbf-, 2020-09-26:
hmm, interesting issue... it's indeed not OK to quit the program when one connection fails.
Should be not so difficult to fix this in the `ethInitConnectionMulti()` function

The V3.7.0 compile issue is explained [here](https://github.com/SBFspot/SBFspot/issues/394#issuecomment-651307208)

Q: did you reset (power down) the faulty inverter already. Maybe this helps?

cav-oem, 2020-10-07:
> hmm, interesting issue... it's indeed not OK to quit the program when one connection fails.
> Should be not so difficult to fix this in the `ethInitConnectionMulti()` function
>
> The V3.7.0 compile [...]

Hi,
The inverter actually reset itself a couple of days ago after a phase drop during the night and the communication actually restored.

I'm guessing that in the `ethInitConnectionMulti()` function, at the if/else, instead of the `return E_INIT;` you would replace it with a `continue` of sorts, to "skip" to the next index of the for loop?

Thanks again!

commit 8735775b (2021-01-05): Fix #412 skipping unresponsive inverter
closed 2021-01-05 by sbf-