SBFspot/SBFspot #414: Some modernization

mincequi, 2020-10-17:
Hi,
thanks for your project and your contribution to the open source community.

Actually, i plan to use SBFspot in a personal project, but i want to add some substantial features which will result in also doing some technical changes.
Currently the planned changes are:
1. CMake support (to replace Makefile and Visual Studio project files with one single build system).
2. Introduce Importer (Network, Bluetooth) and Exporter (CSV, SQL, MQTT) base classes to have a common interface for any data source and sink.
3. Add Bluetooth Low Energy support (to publish the data via BLE to e.g. an iOS/Android App).
4. Make the SBFspot executable a daemon itself (without the need for a cronjob).
5. Create debian package scripts.
6. Introduce a more class-based design
7. Make use of C++11 features (like std::scoped_ptr<> or range based for loops).

My question now is: would you be eager to integrate some/all of that stuff into SBFSpot, when i create pull requests? If that's not the case, i would start a complete new fork/project to realize my requirements. However, i would prefer to contribute to SBFspot to avoid unnecessary splitting off.

What do you think?

sbf-, 2020-10-22:
Not so sure if it's worth the effort to make these drastic changes as described in 2/6/7
All of the modern inverters support Sunny Portal and Modbus. SMAData2+ is an old protocol and there is a big chance it won't be supported in the next generation of inverters. That's the reason why development is at low priority ATM
For the other items:
CMake: nice to have (Not aware of somebody building his own windows executable)
Bluetooth LE: No idea
SBFspot as a daemon: See #318 - Also need for Windows service
Debian package: Several persons promised to help with this, but until now... no results

mincequi, 2020-10-22:
Ok, I see your points. Well, I can do some pull requests. The question is rather: are you willing to integrate that, if I do the work? I try to provide comprehensible steps and not all the points listed above are high priority.

For 1 there is a pull request already (also with some beautifications/clean up). Is this change ok for you?

closed 2021-04-12 by mincequi