SBFspot/SBFspot #417: MQTT publishing issue

Robindeko, 2020-11-08:
SBSpot is working great, but I tried MQTT to connect it to Home Assistant.
It seems that MQTT is using the same values for UAC1, UAC2 and UAC3.
The same goes for values IAC1, IAC2 and IAC3

my log shows different values for these, but MQTT is sending the info regarding L1 only:

AC Spot Data:
	Phase 1 Pac :   0.582kW - **_**_**_Uac: 234.36V_** - Iac:  2.484A_**_**
	Phase 2 Pac :   0.586kW - **_****__Uac: 236.53V_** - Iac:  2.478A_**_**
	Phase 3 Pac :   0.586kW - **_Uac: 236.57V - Iac:  2.479A_**
	Total Pac   :   1.754kW - Calculated Pac:   1.754kW
	Efficiency  :  100.00%
SUSyID: 181 - SN: 304625475
Grid Freq. : 49.99Hz
SUSyID: 181 - SN: 304625475
Current Inverter Time: 08/11/2020 13:03:43
Inverter Wake-Up Time: 08/11/2020 07:54:55
Inverter Sleep Time  : 08/11/2020 13:03:46

MQTT: Publishing (sbfspot_{serial}) "Timestamp": "08/11/2020 13:03:42","SunRise": "08/11/2020 07:51:00","SunSet": "08/11/2020 16:58:00","InvTime": "08/11/2020 13:03:43","InvStatus": "Ok","InvTemperature": 50.610,"InvGridRelay": "Closed","EToday": 4.877,"ETotal": 49331.665,"PACTot": 1754.000,"UDC1": 320.580,"UDC2": 185.310,"IDC1": 3.501,"IDC2": 3.423,"PDC1": 1121.000,"PDC2": 633.000,**"UAC1": 234.360,"UAC2": 234.360,"UAC3": 234.360**,"IAC1": 2.484,"IAC2": 2.484,"IAC3": 2.484,"FeedTm": 32299.037,"OperTm": 33081.696,"PAC1": 582.000,"PAC2": 582.000,"PAC3": 582.000,"GridFreq": 49.990,"PACTot": 1754.000

I'm using Version 3.7.0

sbf-, 2020-11-09:
I checked the code and it's indeed a lazy copy/paste bug
I'll fix it asap

Robindeko, 2020-11-09:
Great, thanks!

Robindeko, 2020-11-10:
I just found out the same goes for PAC1, PAC2 and PAC3,
It's good to see your consistent :D

commit ebe6a479 (2021-01-05): Fix #417: MQTT publishing issue
closed 2021-01-05 by sbf-